saki4510t/UVCCamera #232: COM.SERENEGIANT CAN NOT RENAME AGAIN, THERE ARE A LOCK INSIDE THE PROGRAM.

aulker62, 2017-09-11:
I want to combine UVCCamera and my own project together. I tried more than one month, but there is no any way. I tried refactor package name many times, i tried also rename all files one bye one manually. 
I think so, Mr.Saki putted in the program a LOCK.
You can change the package name but will be error at the end "CAN NOT FIND COMMON-1.5.20." So after the rename, program don't work.

I want to ask, can we use your program program or not?
Thanks...

lb5160482, 2017-09-22:
I actually successfully renamed the library. How did you change the package name?
If you are facing  "CAN NOT FIND COMMON-1.5.20." this problem, remember to add
```
allprojects {
    repositories {
		maven { url 'http://raw.github.com/saki4510t/libcommon/master/repository/' }
        jcenter()
    }
}
```
in your project's main build.gradle.

RogerHardiman, 2017-09-23:
@aulker62   can you tell me why you need to change the package name please?
I combine UVCCamera with my own projects and the whole project uses my APK name (I did not have to rename anything). Don't forget that if you rename the package, you need to rename parts of the C code too (due to JNI security checks)

aulker62, 2017-09-24:
Thanks Roger.

Yes, i combined my own project and uvccamera under my package's name.