saki4510t/UVCCamera #233: Can I keep the camera working without setting the Surface(hide UI)?

lb5160482, 2017-09-14:
If I want to hide the image view and do some image processing(I do not want the UI part), how can I make the camera working?
I tried removing the Surface, then the camera was not working.

sikrinick, 2017-09-15:
Are you using based on UVCCamera or UVCCameraHandler class?

lb5160482, 2017-09-18:
@sikrinick either is fine. I am now using UVCCameraHandler class.

sappho192, 2017-09-18:
Early I faced this problem and solved it like setting preview UI size as 0px * 0px (width * height)? 1px * 1px?

lb5160482, 2017-09-18:
@sappho192 Thanks! Yeah I tried this before and it worked. What  I am facing now is I am going to build a camera plugin for Unity. In that way, the UI is generated by Unity, where probably I have no access to the XML layout file. I can manually add the surface component on Java side and then set the size 1 pixel. But I am wondering if I can totally get rid of it?

xiaochongdeweiba, 2017-09-28:
usbcameratest4 can do this

lb5160482, 2017-09-29:
Solved by using [here](https://github.com/TimGroeneboom/libuvc_android_unity).

closed 2017-09-29 by lb5160482

MarcBS, 2020-10-20:
@lb5160482 it seems that the link you provided as a solution is no longer available. Could you please share an updated link or a snippet? 

That would be very helpful!