saki4510t/UVCCamera #234: Crash when open the camera

xiongyix, 2017-09-16:
Hey.
I run the usbCameraTest app.
After app is started, click the camera button, choose the uvc device, then the app crash. I got the following logs.
Thanks for helping me to find the error.

09-17 01:18:59.681 999-999/? I/art: Late-enabling -Xcheck:jni
09-17 01:18:59.741 999-999/com.serenegiant.usbcameratest W/ActivityThread: Application com.serenegiant.usbcameratest is waiting for the debugger on port 8100...
09-17 01:18:59.745 999-999/com.serenegiant.usbcameratest I/System.out: Sending WAIT chunk
09-17 01:18:59.911 999-1006/com.serenegiant.usbcameratest I/art: Debugger is active
09-17 01:18:59.945 999-999/com.serenegiant.usbcameratest I/System.out: Debugger has connected
09-17 01:18:59.946 999-999/com.serenegiant.usbcameratest I/System.out: waiting for debugger to settle...
09-17 01:19:00.146 999-999/com.serenegiant.usbcameratest I/System.out: waiting for debugger to settle...
09-17 01:19:00.346 999-999/com.serenegiant.usbcameratest I/System.out: waiting for debugger to settle...
09-17 01:19:00.546 999-999/com.serenegiant.usbcameratest I/System.out: waiting for debugger to settle...
09-17 01:19:00.747 999-999/com.serenegiant.usbcameratest I/System.out: waiting for debugger to settle...
09-17 01:19:00.947 999-999/com.serenegiant.usbcameratest I/System.out: waiting for debugger to settle...
09-17 01:19:01.147 999-999/com.serenegiant.usbcameratest I/System.out: waiting for debugger to settle...
09-17 01:19:01.348 999-999/com.serenegiant.usbcameratest I/System.out: waiting for debugger to settle...
09-17 01:19:01.548 999-999/com.serenegiant.usbcameratest I/System.out: waiting for debugger to settle...
09-17 01:19:01.748 999-999/com.serenegiant.usbcameratest I/System.out: debugger has settled (1458)
09-17 01:19:01.859 999-999/com.serenegiant.usbcameratest W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --instruction-set-features=smp,div,-atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a7 --instruction-set-features=default --dex-file=/data/app/com.serenegiant.usbcameratest-2/split_lib_dependencies_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.serenegiant.usbcameratest-2@split_lib_dependencies_apk.apk@classes.dex) because non-0 exit status
09-17 01:19:02.130 999-999/com.serenegiant.usbcameratest W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --instruction-set-features=smp,div,-atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a7 --instruction-set-features=default --dex-file=/data/app/com.serenegiant.usbcameratest-2/split_lib_slice_0_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.serenegiant.usbcameratest-2@split_lib_slice_0_apk.apk@classes.dex) because non-0 exit status
09-17 01:19:02.196 999-999/com.serenegiant.usbcameratest W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --instruction-set-features=smp,div,-atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a7 --instruction-set-features=default --dex-file=/data/app/com.serenegiant.usbcameratest-2/split_lib_slice_1_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.serenegiant.usbcameratest-2@split_lib_slice_1_apk.apk@classes.dex) because non-0 exit status
09-17 01:19:02.273 999-999/com.serenegiant.usbcameratest W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --instruction-set-features=smp,div,-atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a7 --instruction-set-features=default --dex-file=/data/app/com.serenegiant.usbcameratest-2/split_lib_slice_2_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.serenegiant.usbcameratest-2@split_lib_slice_2_apk.apk@classes.dex) because non-0 exit status
09-17 01:19:02.347 999-999/com.serenegiant.usbcameratest W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --instruction-set-features=smp,div,-atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a7 --instruction-set-features=default --dex-file=/data/app/com.serenegiant.usbcameratest-2/split_lib_slice_3_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.serenegiant.usbcameratest-2@split_lib_slice_3_apk.apk@classes.dex) because non-0 exit status
09-17 01:19:02.412 999-999/com.serenegiant.usbcameratest W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --instruction-set-features=smp,div,-atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a7 --instruction-set-features=default --dex-file=/data/app/com.serenegiant.usbcameratest-2/split_lib_slice_4_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.serenegiant.usbcameratest-2@split_lib_slice_4_apk.apk@classes.dex) because non-0 exit status
09-17 01:19:02.476 999-999/com.serenegiant.usbcameratest W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --instruction-set-features=smp,div,-atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a7 --instruction-set-features=default --dex-file=/data/app/com.serenegiant.usbcameratest-2/split_lib_slice_5_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.serenegiant.usbcameratest-2@split_lib_slice_5_apk.apk@classes.dex) because non-0 exit status
09-17 01:19:02.547 999-999/com.serenegiant.usbcameratest W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --instruction-set-features=smp,div,-atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a7 --instruction-set-features=default --dex-file=/data/app/com.serenegiant.usbcameratest-2/split_lib_slice_6_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.serenegiant.usbcameratest-2@split_lib_slice_6_apk.apk@classes.dex) because non-0 exit status
09-17 01:19:02.624 999-999/com.serenegiant.usbcameratest W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --instruction-set-features=smp,div,-atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a7 --instruction-set-features=default --dex-file=/data/app/com.serenegiant.usbcameratest-2/split_lib_slice_7_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.serenegiant.usbcameratest-2@split_lib_slice_7_apk.apk@classes.dex) because non-0 exit status
09-17 01:19:02.695 999-999/com.serenegiant.usbcameratest W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --instruction-set-features=smp,div,-atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a7 --instruction-set-features=default --dex-file=/data/app/com.serenegiant.usbcameratest-2/split_lib_slice_8_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.serenegiant.usbcameratest-2@split_lib_slice_8_apk.apk@classes.dex) because non-0 exit status
09-17 01:19:02.765 999-999/com.serenegiant.usbcameratest W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --instruction-set-features=smp,div,-atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a7 --instruction-set-features=default --dex-file=/data/app/com.serenegiant.usbcameratest-2/split_lib_slice_9_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.serenegiant.usbcameratest-2@split_lib_slice_9_apk.apk@classes.dex) because non-0 exit status
[... cut ...]
09-17 01:22:00.227 999-1108/com.serenegiant.usbcameratest I/libusb/core: [1108*core.c:2085:libusb_init2]:libusb v1.0.19.10903
09-17 01:22:00.228 999-1108/com.serenegiant.usbcameratest I/libusb/usbfs: [1108*android_usbfs.c:504:op_init2]:start up hotplug event handler
09-17 01:22:00.228 999-1108/com.serenegiant.usbcameratest I/libusb/usbfs: [1108*android_usbfs.c:512:op_init2]:call android_scan_devices
09-17 01:22:00.232 999-1108/com.serenegiant.usbcameratest A/libc: Fatal signal 11 (SIGSEGV), code 1, fault addr 0x6f0043 in tid 1108 (BaseActivity)

htoomyintnaung, 2017-10-11:
I'm also having same problem. Help please.

knat, 2017-10-23:
open libuvccamera\src\main\jni\Application.mk, uncomment
```
#NDK_TOOLCHAIN_VERSION := 4.9
```
rebuild