saki4510t/UVCCamera #238: Build Unity Android Plugin

lb5160482, 2017-09-23:
Anyone successfully built the libuvccamera and usbCameraCommon as Unity Android plugin?
I am building an Unity Android plugin which has dependency of the uvccamera library. I have already set up the configurations correctly in Android Studio so it is able to find the functions from uvccamera library. After I build the .jar file(i tried .aar as well) and run my Unity app, it crashes with the problem:
```
java.lang.ClassNotFoundException: com.serenegiant.widget.UVCCameraTextureView
java.lang.ClassNotFoundException: Didn't find class "com.serenegiant.widget.IAspectRatioView" on path...
```
For this error, this explanation seems to make sense: 

>[ "This is caused when there is a class file that your code depends on and it is present at compile time but not found at runtime"](https://stackoverflow.com/questions/34413/why-am-i-getting-a-noclassdeffounderror-in-java)

I then build libuvccamera and usbCameraCommon, put the two .jar files and the .so files under my Unity Plugin folder. Here is the structure:
```
\Assets\Plugins\Android\jni\armeabi-v7a\libjpeg-turbo1500.so
\Assets\Plugins\Android\jni\armeabi-v7a\libusb100.so
\Assets\Plugins\Android\jni\armeabi-v7a\libuvc.so
\Assets\Plugins\Android\jni\armeabi-v7a\libUVCCamera.so
\Assets\Plugins\Android\libs\libuvccamera.jar
\Assets\Plugins\Android\libs\usbCameraCommon.jar
\Assets\Plugins\Android\myplugin.jar
```
However, it keeps crashing.
I also tried putting all the .so files under Android folder( this works for OpenCV), but same results.
Any one can help to figure out how to build this library into Unity Android plugin. 
Thanks!

lb5160482, 2017-09-29:
Solved by [here](https://github.com/TimGroeneboom/libuvc_android_unity)

closed 2017-09-29 by lb5160482

fanzhouyou, 2020-06-23:
404

fanzhouyou, 2020-06-23:
> Solved by [here](https://github.com/TimGroeneboom/libuvc_android_unity)

404

zipswich, 2022-04-20:
Add the following to settings.gradle:
`maven { url 'https://raw.github.com/saki4510t/libcommon/master/repository/' }`

Add the following to build.gradle:
```
    implementation("com.serenegiant:common:4.1.1") {
        exclude module: 'support-v4'
    }
```