saki4510t/UVCCamera #239: Many classes are missing when trying to capture still image

amani-mohammad, 2017-09-26:
I have downloaded all library files and tried Test, Test0, Test1 and Test2 and they all worked fine. But when I tried to run Test3 - which contains the functionality of taking still images -, I found many errors regarding missing classes.

The missing classes are: FpsCounter, IUVCServiceOnFrameAvailable, GLDrawer2D, EGLBase, IAspectRatioView, RendererHolder and RenderHandler.

I have searched for them in common directory and also in all tests folders, but didn't find any thing.

What is missing ? and how can I find them ?

xiaochongdeweiba, 2017-09-28:
you can use Android Studio , "GoTo"  "Declaration" than you can get all this file.

xiaochongdeweiba, 2017-09-28:
you can use Android Studio , "GoTo" "Declaration" than you can get all these files.