saki4510t/UVCCamera #240: IFrameCallback is not work

zhbhhh, 2017-09-28:
I did setFrameCallback and setPreviewDisplay, but I get nothing on onFrame.why?

UbuntuLover, 2017-09-29:
try paste some logs on this page.

zhbhhh, 2017-09-29:
this is my preview function:
public void handleStartPreview(final int width, final int height, final Surface surface) {
			if (DEBUG) Log.d(TAG_THREAD, "handleStartPreview:");
			synchronized (mSync) {
				if (mUVCCamera == null) return;
				try {
//					mUVCCamera.setPreviewSize(width, height, UVCCamera.PIXEL_FORMAT_RGB565);
					mUVCCamera.setPreviewSize(width, height);
				} catch (final IllegalArgumentException e) {
					try {
						// fallback to YUV mode
						Toast.makeText(mContext,"fail to yuv mode",Toast.LENGTH_SHORT).show();
						mUVCCamera.setPreviewSize(width, height, UVCCamera.DEFAULT_PREVIEW_MODE);
					} catch (final IllegalArgumentException e1) {
						mUVCCamera.destroy();
						mUVCCamera = null;
					}
				}
				Toast.makeText(mContext,"startPreView",Toast.LENGTH_SHORT).show();
				mFrameWidth = width;
				mFrameHeight = height;
				mUVCCamera.setPreviewDisplay(surface);
				mUVCCamera.startPreview();
				mUVCCamera.setFrameCallback(mIFrameCallback, UVCCamera.PIXEL_FORMAT_YUV);
			}
		}

this is my callback:
private final IFrameCallback mIFrameCallback = new IFrameCallback() {
			@Override
			public void onFrame(final ByteBuffer frame) {
				byte[] array = frame.array();
				Toast.makeText(mContext,"kjhkjhkjhkjh",Toast.LENGTH_SHORT).show();
				showToast("jhjhhhhh");
			}
		};

but the function onFrame has not executed,

lb5160482, 2017-10-02:
You need to call open, setPreviewSize, setPreviewTexture and then startPreview to receive camera images.

Levi-ackerman, 2017-10-03:
@lb5160482 I also open, setPreviewSize, setPreviewTexture, startPreview but I get nothing in onFrame callback ?

zhbhhh, 2017-10-09:
@lb5160482 I did all,but I dont receive camera callback data.

zhbhhh, 2017-10-09:
@Levi-ackerman Did you solve it?

Levi-ackerman, 2017-10-09:
@zhbhhh Yes, i have receive frames but if I log it in current thread, nothing to show, should log it in main thread.

zhbhhh, 2017-10-09:
@Levi-ackerman Can I have a look at your code?
This is my code:
// if you need frame data as ByteBuffer on Java side, you can use this callback method with UVCCamera#setFrameCallback
		private final IFrameCallback mIFrameCallback = new IFrameCallback() {
			@Override
			public void onFrame(final ByteBuffer frame) {
				byte[] array = frame.array();
				Toast.makeText(mContext,"kjhkjhkjhkjh",Toast.LENGTH_SHORT).show();
				showToast("jhjhhhhh");
			}
		};

		private void showToast(final String text) {
			((Activity) mContext).runOnUiThread(new Runnable() {
				@Override
				public void run() {
					Toast.makeText(mContext, text, Toast.LENGTH_LONG).show();
				}
			});
		}

Levi-ackerman, 2017-10-09:
@zhbhhh You need convert ByteBuffer to byte array. Use this code : 

`public void onFrame(final ByteBuffer frame) { `
    `   byte[] arr = new byte[frame.remaining()];`
   `    frame.get(arr);`
`}`

zhbhhh, 2017-10-09:
@Levi-ackerman The problem is that my method onFrame is not implemented.I can preview,but I cant receive frames.

lb5160482, 2017-10-09:
@zhbhhh Did you have setFrameCallback() implemented?

closed 2017-10-17 by zhbhhh

xchangfeng, 2018-07-28:
@zhbhhh Do you solve the problem? If solved, can you tell  how?

1159658610, 2018-07-28:
I also can preview,but I can't receive frames.

UbuntuLover, 2018-07-30:
@zhbhhh 是的。考虑一下颜色格式设置的是否不对？

UbuntuLover, 2018-07-30:
我设置的是YUV420SP