saki4510t/UVCCamera #241: About getting frame data！！！ Please help me！！！

junbomMagnus, 2017-09-28:
I want to get the frame data, but maybe my method is wrong. Can someone please help me to see it?

My Code：
` private int buf_sz = 4096;
        private byte[] buf = new byte[buf_sz];
        @Override
        public void onConnect(final UsbDevice device, final USBMonitor.UsbControlBlock ctrlBlock, final boolean createNew) {
//                showToast("onConnect:"+ctrlBlock);

            mCameraHandler.open(ctrlBlock);
                final SurfaceTexture st = mUVCCameraView.getSurfaceTexture();
                mCameraHandler.startPreview(new Surface(st));
                pEncoder.onStop(false);
                mUVCCamera = new UVCCamera();
                mIFrameCallback = new IFrameCallback() {
                    @Override
                    public void onFrame(ByteBuffer frame) {

                        final int n = frame.limit();
                        if (buf_sz < n) {
                            buf_sz = n;
                            buf = new byte[n];
                        }
                        frame.get(buf, 0, n);
                         Log.e(TAG,"data::"+buf);
                        pEncoder.encode(buf);
                    }
                };
            mUVCCamera.setFrameCallback(mIFrameCallback, UVCCamera.PIXEL_FORMAT_NV21);
            pEncoder.onStart();
        }`

It's important to me，thank you everybody!

junbomMagnus, 2017-09-29:
The problem has already been solved.

closed 2017-09-29 by junbomMagnus

ducanh841988, 2017-10-02:
Can you tell me how you solve this problem? @junbomMagnus 

Thanks.

junbomMagnus, 2017-10-13:
@ducanh841988 
AbstractUVCCameraHandler.java
handleMessage() add->case MSG_GET_FRAME:
			thread.getFrameCallBack((IFrameCallback) msg.obj);
				break;
add ->public void getFrameCallBack(IFrameCallback iIFrameCallback)
		{
			mUVCCamera.setFrameCallback(iIFrameCallback, UVCCamera.PIXEL_FORMAT_YUV420SP);
		}
UVCCameraHandler.java
add->public void getFrame(IFrameCallback iFrameCallback)
	{
		super.getFrame(iFrameCallback);
	}

use---->
                    mIFrameCallback = new IFrameCallback() {
                        @Override
                        public void onFrame(ByteBuffer frame) {
                            Log.e(TAG,"data::"+buf);
                            final int n = frame.limit();
                            if (buf_sz < n) {
                                buf_sz = n;
                                buf = new byte[n];
                            }
                            frame.get(buf, 0, n);
                             Log.e(TAG,"data::"+buf);
                            pEncoder.encode(buf);
                        }
                    };
                mCameraHandler.getFrame(mIFrameCallback);

This is the way I can do it. I hope I can help you.

ManojMMWWIndia, 2017-10-18:
byte data that we get here does it content video+audio of stream?

junbomMagnus, 2017-10-19:
@ManojMMWWIndia 
 Only video stream part.Audio part, you can write it yourself or refer to his recordings class.

shaniceweiwei, 2018-01-02:
@ManojMMWWIndia 
UVCCameraHandler.java has no super.getFrame(iFrameCallback);
and where does MSG_GET_FRAME comes from?

jeranio, 2022-01-26:
I have implemented this to the letter but the IFramecallback is never triggered. Any suggestions?