saki4510t/UVCCamera #243: UVCCamera7不能同时录像。

UbuntuLover, 2017-10-09:
同时开启录像时，两个都可以录像但是实际上只有一个可以生成有效的mp4文件。另外一个总是时长为0. 大家有遇到这样的问题吗

UbuntuLover, 2017-10-09:
@wf9a5m75 @saki4510t hi  when i try to record the video both 2 cameras using the UVCCameraTest7, i only could get one real mp4 file while other one is invalidate. How can i solve this issue?

zhbhhh, 2017-10-09:
请问 您的DEMO能跑起来吗？我的下下来少了很多包，跑不起来。

UbuntuLover, 2017-10-10:
@zhbhhh 可以啊 这里说一句  编译要用NDK-r13b编译 不然编译通过却捕捉不到画面。

jp1017, 2017-10-10:
@UbuntuLover 我的两个同时预览都不行,一个提示带宽不足,,,r14b编译的

UbuntuLover, 2017-10-10:
@jp1017 你换r13b编译吧 我也是踩了不少坑

jp1017, 2017-10-10:
@UbuntuLover 好的, 多谢提供思路 :smile: :+1:

UbuntuLover, 2017-10-10:
你试试看test7录像有没有声音 在线等 @jp1017

jp1017, 2017-10-10:
@UbuntuLover 没有的,你看看test7的MainActivity前面的注释....

zhbhhh, 2017-10-10:
@UbuntuLover 

import com.serenegiant.dialog.MessageDialogFragment;
import com.serenegiant.utils.BuildCheck;
import com.serenegiant.utils.HandlerThreadHandler;
import com.serenegiant.utils.PermissionCheck;

我的少这些包，是怎么回事儿呢？

UbuntuLover, 2017-10-10:
@jp1017  我只有一个画框可以录像，不能两个一起录像。并且不清楚有没有声音。

UbuntuLover, 2017-10-10:
@zhbhhh 你要么没有下载完整 要么编译出错了。你SDK用最新的版本编译。然后NDK你手动指定为r13b。慢慢来 不要急

zhbhhh, 2017-10-10:
@UbuntuLover 我下下来的都找不到这些包，你那里有这些包吗？源码点进去也没有这些包，所以有点奇怪。

UbuntuLover, 2017-10-10:
@zhbhhh  我qq 1091211092

closed 2017-10-11 by UbuntuLover