saki4510t/UVCCamera #244: JNI DETECTED ERROR IN APPLICATION: jmethodID was NULL

xiaozhi003, 2017-10-09:
```java
01-19 00:07:56.883 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/java_vm_ext.cc:470] JNI DETECTED ERROR IN APPLICATION: jmethodID was NULL
01-19 00:07:56.883 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/java_vm_ext.cc:470]     in call to CallVoidMethodV
01-19 00:07:56.883 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/java_vm_ext.cc:470] "Thread-75" prio=5 tid=22 Runnable
01-19 00:07:56.883 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/java_vm_ext.cc:470]   | group="main" sCount=0 dsCount=0 obj=0x12f94a60 self=0x76d602b600
01-19 00:07:56.883 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/java_vm_ext.cc:470]   | sysTid=19466 nice=0 cgrp=default sched=0/0 handle=0x76c1784450
01-19 00:07:56.883 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/java_vm_ext.cc:470]   | state=R schedstat=( 117810374 90318833 156 ) utm=10 stm=1 core=3 HZ=100
01-19 00:07:56.883 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/java_vm_ext.cc:470]   | stack=0x76c168a000-0x76c168c000 stackSize=1005KB
01-19 00:07:56.883 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/java_vm_ext.cc:470]   | held mutexes= "mutator lock"(shared held)
01-19 00:07:56.883 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/java_vm_ext.cc:470]   native: #00 pc 000000000047ef3c  /system/lib64/libart.so (_ZN3art15DumpNativeStackERNSt3__113basic_ostreamIcNS0_11char_traitsIcEEEEiP12BacktraceMapPKcPNS_9ArtMethodEPv+220)
01-19 00:07:56.883 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/java_vm_ext.cc:470]   native: #01 pc 000000000047ef38  /system/lib64/libart.so (_ZN3art15DumpNativeStackERNSt3__113basic_ostreamIcNS0_11char_traitsIcEEEEiP12BacktraceMapPKcPNS_9ArtMethodEPv+216)
01-19 00:07:56.883 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/java_vm_ext.cc:470]   native: #02 pc 0000000000452fc4  /system/lib64/libart.so (_ZNK3art6Thread9DumpStackERNSt3__113basic_ostreamIcNS1_11char_traitsIcEEEEbP12BacktraceMap+480)
01-19 00:07:56.883 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/java_vm_ext.cc:470]   native: #03 pc 00000000002f04a4  /system/lib64/libart.so (_ZN3art9JavaVMExt8JniAbortEPKcS2_+1136)
01-19 00:07:56.883 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/java_vm_ext.cc:470]   native: #04 pc 00000000002f0bb4  /system/lib64/libart.so (_ZN3art9JavaVMExt9JniAbortVEPKcS2_St9__va_list+124)
01-19 00:07:56.883 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/java_vm_ext.cc:470]   native: #05 pc 0000000000102798  /system/lib64/libart.so (_ZN3art11ScopedCheck6AbortFEPKcz+156)
01-19 00:07:56.883 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/java_vm_ext.cc:470]   native: #06 pc 0000000000101f60  /system/lib64/libart.so (_ZN3art11ScopedCheck13CheckMethodIDERNS_18ScopedObjectAccessEP10_jmethodID+180)
01-19 00:07:56.883 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/java_vm_ext.cc:470]   native: #07 pc 0000000000100d6c  /system/lib64/libart.so (_ZN3art11ScopedCheck22CheckPossibleHeapValueERNS_18ScopedObjectAccessEcNS_12JniValueTypeE+92)
01-19 00:07:56.883 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/java_vm_ext.cc:470]   native: #08 pc 00000000000ffce8  /system/lib64/libart.so (_ZN3art11ScopedCheck5CheckERNS_18ScopedObjectAccessEbPKcPNS_12JniValueTypeE+1124)
01-19 00:07:56.883 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/java_vm_ext.cc:470]   native: #09 pc 0000000000105664  /system/lib64/libart.so (_ZN3art8CheckJNI13CheckCallArgsERNS_18ScopedObjectAccessERNS_11ScopedCheckEP7_JNIEnvP8_jobjectP7_jclassP10_jmethodIDNS_10InvokeTypeEPKNS_7VarArgsE+140)
01-19 00:07:56.883 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/java_vm_ext.cc:470]   native: #10 pc 0000000000104920  /system/lib64/libart.so (_ZN3art8CheckJNI11CallMethodVEPKcP7_JNIEnvP8_jobjectP7_jclassP10_jmethodIDSt9__va_listNS_9Primitive4TypeENS_10InvokeTypeE+704)
01-19 00:07:56.883 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/java_vm_ext.cc:470]   native: #11 pc 00000000000f76d8  /system/lib64/libart.so (_ZN3art8CheckJNI15CallVoidMethodVEP7_JNIEnvP8_jobjectP10_jmethodIDSt9__va_list+96)
01-19 00:07:56.883 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/java_vm_ext.cc:470]   native: #12 pc 0000000000013ec4  /data/app/com.serenegiant.usbcameratest3-2/lib/arm64/libUVCCamera.so (_ZN7_JNIEnv14CallVoidMethodEP8_jobjectP10_jmethodIDz+120)
01-19 00:07:56.883 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/java_vm_ext.cc:470]   native: #13 pc 0000000000013d14  /data/app/com.serenegiant.usbcameratest3-2/lib/arm64/libUVCCamera.so (_ZN10UVCPreview19do_capture_callbackEP7_JNIEnvP9uvc_frame+188)
01-19 00:07:56.883 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/java_vm_ext.cc:470]   native: #14 pc 00000000000139ec  /data/app/com.serenegiant.usbcameratest3-2/lib/arm64/libUVCCamera.so (_ZN10UVCPreview10do_captureEP7_JNIEnv+108)
01-19 00:07:56.883 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/java_vm_ext.cc:470]   native: #15 pc 00000000000133ec  /data/app/com.serenegiant.usbcameratest3-2/lib/arm64/libUVCCamera.so (_ZN10UVCPreview19capture_thread_funcEPv+60)
01-19 00:07:56.884 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/java_vm_ext.cc:470]   native: #16 pc 0000000000068748  /system/lib64/libc.so (_ZL15__pthread_startPv+208)
01-19 00:07:56.884 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/java_vm_ext.cc:470]   native: #17 pc 000000000001da7c  /system/lib64/libc.so (__start_thread+16)
01-19 00:07:56.884 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/java_vm_ext.cc:470]   (no managed stack frames)
01-19 00:07:56.884 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/java_vm_ext.cc:470] 
01-19 00:07:56.990 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/runtime.cc:422] Runtime aborting...
01-19 00:07:56.990 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/runtime.cc:422] Aborting thread:
01-19 00:07:56.990 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/runtime.cc:422] "Thread-75" prio=5 tid=22 Native
01-19 00:07:56.990 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/runtime.cc:422]   | group="" sCount=0 dsCount=0 obj=0x12f94a60 self=0x76d602b600
01-19 00:07:56.990 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/runtime.cc:422]   | sysTid=19466 nice=0 cgrp=default sched=0/0 handle=0x76c1784450
01-19 00:07:56.990 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/runtime.cc:422]   | state=R schedstat=( 146805837 100435289 186 ) utm=11 stm=2 core=3 HZ=100
01-19 00:07:56.990 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/runtime.cc:422]   | stack=0x76c168a000-0x76c168c000 stackSize=1005KB
01-19 00:07:56.990 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/runtime.cc:422]   | held mutexes= "abort lock"
01-19 00:07:56.990 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/runtime.cc:422]   native: #00 pc 000000000047ef3c  /system/lib64/libart.so (_ZN3art15DumpNativeStackERNSt3__113basic_ostreamIcNS0_11char_traitsIcEEEEiP12BacktraceMapPKcPNS_9ArtMethodEPv+220)
01-19 00:07:56.990 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/runtime.cc:422]   native: #01 pc 000000000047ef38  /system/lib64/libart.so (_ZN3art15DumpNativeStackERNSt3__113basic_ostreamIcNS0_11char_traitsIcEEEEiP12BacktraceMapPKcPNS_9ArtMethodEPv+216)
01-19 00:07:56.990 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/runtime.cc:422]   native: #02 pc 0000000000452fc4  /system/lib64/libart.so (_ZNK3art6Thread9DumpStackERNSt3__113basic_ostreamIcNS1_11char_traitsIcEEEEbP12BacktraceMap+480)
01-19 00:07:56.990 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/runtime.cc:422]   native: #03 pc 0000000000440f3c  /system/lib64/libart.so (_ZNK3art10AbortState10DumpThreadERNSt3__113basic_ostreamIcNS1_11char_traitsIcEEEEPNS_6ThreadE+56)
01-19 00:07:56.991 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/runtime.cc:422]   native: #04 pc 0000000000440db8  /system/lib64/libart.so (_ZNK3art10AbortState4DumpERNSt3__113basic_ostreamIcNS1_11char_traitsIcEEEE+668)
01-19 00:07:56.991 16786-19466/com.serenegiant.usbcameratest3 A/art: art/runtime/runtime.cc:422]   native: #05 pc 000000000043473c  /system/lib64/libart.so (_ZN3art7Runtime5AbortEPKc+148)
[... cut ...]
01-19 00:07:57.092 19512-19512/? A/DEBUG:     #17 pc 00000000000139ec  /data/app/com.serenegiant.usbcameratest3-2/lib/arm64/libUVCCamera.so (_ZN10UVCPreview10do_captureEP7_JNIEnv+108)
01-19 00:07:57.092 19512-19512/? A/DEBUG:     #18 pc 00000000000133ec  /data/app/com.serenegiant.usbcameratest3-2/lib/arm64/libUVCCamera.so (_ZN10UVCPreview19capture_thread_funcEPv+60)
01-19 00:07:57.092 19512-19512/? A/DEBUG:     #19 pc 0000000000068748  /system/lib64/libc.so (_ZL15__pthread_startPv+208)
01-19 00:07:57.092 19512-19512/? A/DEBUG:     #20 pc 000000000001da7c  /system/lib64/libc.so (__start_thread+16)
```

SunPointed, 2018-06-13:
hi @xiaozhi003 , I meet this  problem(JNI DETECTED ERROR IN APPLICATION: jmethodID was NULL  in call to CallVoidMethodV). I already see #259 and a rebuild so libraries, but when I called UVCCamera's close() method, which randomly occur this problem when close() method called stopPreview(). Have you solved the problem， can you give me some advice? 如果可以的话，可以用中文交流

xiaozhi003, 2018-06-15:
首先要确认几点问题：
1.修改完之后，多次打开内存是否稳定不增长了
2.使用的Android板子是什么型号的
3.摄像头是什么型号的，是单开还是双开
4.具体打开多少次会出现问题
5.使用的是demo几

据我所知，我是删除了usbCameraCommon，不使用UVCCameraHandler操作摄像头，精简了libuvccamera moudle，删除了录制视频的代码等。

除了项目本身的问题，还有好多问题来自于Android开发板usb驱动的问题等等

SunPointed, 2018-06-15:
[image]
[image]
`void UVCPreview::do_capture_callback(JNIEnv *env, uvc_frame_t *frame) {
	ENTER();

	if (LIKELY(frame)) {
		uvc_frame_t *callback_frame = frame;
		if (mFrameCallbackObj) {
			if (mFrameCallbackFunc) {
				callback_frame = get_frame(callbackPixelBytes);
				if (LIKELY(callback_frame)) {
					int b = mFrameCallbackFunc(frame, callback_frame);
					recycle_frame(frame);
					if (UNLIKELY(b)) {
						LOGW("failed to convert for callback frame");
						goto SKIP;
					}
				} else {
					LOGW("failed to allocate for callback frame");
					callback_frame = frame;
					goto SKIP;
				}
			}
			jobject buf = env->NewDirectByteBuffer(callback_frame->data, callbackPixelBytes);
			pthread_mutex_lock(&capture_mutex);
			if(iframecallback_fields.onFrame != NULL){
				env->CallVoidMethod(mFrameCallbackObj, iframecallback_fields.onFrame, buf);
			}
			pthread_mutex_unlock(&capture_mutex);
			env->ExceptionClear();
			env->DeleteLocalRef(buf);
		}
 SKIP:
		recycle_frame(callback_frame);
	}
	EXIT();
}`
估计是c++层设置停止标志位后，preview还可能会执行最后一次，而java层直接把callback（iframecallback_fields.onFrame为NULL）设置为NULL了，就导致了这个JNI DETECTED ERROR IN APPLICATION: jmethodID was NULL

xiaozhi003, 2018-06-15:
哈哈，我也改过这里，改了好多地方有些地方忘记了。这个地方确实需要再判断一下，但是我没有加线程锁

```java
void UVCPreview::do_capture_callback(JNIEnv *env, uvc_frame_t *frame) {
	ENTER();

	if (LIKELY(frame)) {
		uvc_frame_t *callback_frame = frame;
		if (mFrameCallbackObj) {
			if (mFrameCallbackFunc) {
				callback_frame = get_frame(callbackPixelBytes);
				if (LIKELY(callback_frame)) {
					int b = mFrameCallbackFunc(frame, callback_frame);
					recycle_frame(frame);
					if (UNLIKELY(b)) {
						LOGW("failed to convert for callback frame");
						goto SKIP;
					}
				} else {
					LOGW("failed to allocate for callback frame");
					callback_frame = frame;
					goto SKIP;
				}
			}
			jobject buf = env->NewDirectByteBuffer(callback_frame->data, callbackPixelBytes);
			if(iframecallback_fields.onFrame) {
			    env->CallVoidMethod(mFrameCallbackObj, iframecallback_fields.onFrame, buf);
			}
			env->ExceptionClear();
			env->DeleteLocalRef(buf);
		}
 SKIP:
		recycle_frame(callback_frame);
	}
	EXIT();
}
```

SunPointed, 2018-06-15:
不加锁有可能刚好if判断完了，iframecallback_fields.onFrame在另外一个线程被置为NULL

paipeng, 2018-07-06:
大家好，我们发现了相同的问题。出在一款Android特制手机上。

通过简单 处理与上面的解决方案类似，完成并测试通过。问题不再出现。

`void UVCPreview::do_capture_callback(JNIEnv *env, uvc_frame_t *frame) {
	ENTER();

	if (LIKELY(frame)) {
		uvc_frame_t *callback_frame = frame;
		if (mFrameCallbackObj) {
			if (mFrameCallbackFunc) {
				callback_frame = get_frame(callbackPixelBytes);
				if (LIKELY(callback_frame)) {
					int b = mFrameCallbackFunc(frame, callback_frame);
					recycle_frame(frame);
					if (UNLIKELY(b)) {
						LOGW("failed to convert for callback frame");
						goto SKIP;
					}
				} else {
					LOGW("failed to allocate for callback frame");
					callback_frame = frame;
					goto SKIP;
				}
			}
			if (iframecallback_fields.onFrame != NULL) {
			jobject buf = env->NewDirectByteBuffer(callback_frame->data, callbackPixelBytes);
			env->CallVoidMethod(mFrameCallbackObj, iframecallback_fields.onFrame, buf);
			env->ExceptionClear();
			env->DeleteLocalRef(buf);
			}
		}
 SKIP:
		recycle_frame(callback_frame);
	}
	EXIT();
}`

VladimirKulyk, 2018-08-08:
Same problem, native crash on closing camera `nativeStopPreview(mNativePtr);`

hugoleong, 2019-11-15:
public synchronized void stopPreview() {
		//setFrameCallback(null, 0);
    	if (mCtrlBlock != null) {
    		nativeStopPreview(mNativePtr);
    	}
    }
把stopPreView()的setFrameCallback(null, 0); 这句注释掉，就完美解决这个问题了。

楼上的方案：
if (iframecallback_fields.onFrame != NULL)加上这句判断，因为异步的问题，还是有一定概率会发生这个错误。

jjjjbababy, 2020-05-05:
> ```
> public synchronized void stopPreview() {
> 	//setFrameCallback(null, 0);
> 	if (mCtrlBlock != null) {
> 		nativeStopPreview(mNativePtr);
> 	}
> }
> ```
> 
> 把stopPreView()的setFrameCallback(null, 0); 这句注释掉，就完美解决这个问题了。
> 
> 楼上的方案：
> if (iframecallback_fields.onFrame != NULL)加上这句判断，因为异步的问题，还是有一定概率会发生这个错误。

我注释掉了 这行代码，目前没再出现闪退问题，您是这么解决的吗

george-vlahakis, 2020-11-01:
Is this issue still open? A couple of suggestions but I see the ticket still open. Any final thoughts here?

closed 2021-02-08 by xiaozhi003

savet-save, 2023-05-23:
i fix this in multithreading
old: 
```
void CallbackPipeline::do_capture(JNIEnv *env) {
...
if (LIKELY(frame)) {
	uvc_frame_t *callback_frame = frame;
	if (mFrameCallbackObj) {
		if (mFrameCallbackFunc) {
			callback_frame = get_frame(callbackPixelBytes);
			if (LIKELY(callback_frame)) {
				int b = mFrameCallbackFunc(frame, callback_frame);
				recycle_frame(frame);
				if (UNLIKELY(b)) {
					LOGW("failed to convert for callback frame");
					goto SKIP;
				}
			} else {
				LOGW("failed to allocate for callback frame");
				callback_frame = frame;
				goto SKIP;
			}
		}
		jobject buf = env->NewDirectByteBuffer(callback_frame->data, callbackPixelBytes);
		env->CallVoidMethod(mFrameCallbackObj, iframecallback_fields.onFrame, buf);
		env->ExceptionClear();
		env->DeleteLocalRef(buf);
	}
...
```

fix :
```
void CallbackPipeline::do_capture(JNIEnv *env) {
...
	if (LIKELY(temp)) {
		for (; isRunning() && isCapturing();) {
			frame = waitCaptureFrame();
			if ((LIKELY(frame))) {
				if (UNLIKELY((width != frame->width) || (height != frame->height))) {
					width = frame->width;
					height = frame->height;
					callbackPixelFormatChanged(width, height);
					uvc_ensure_frame_size(temp, callbackPixelBytes);
					sz = callbackPixelBytes;
				}
				if (mFrameCallbackObj) {
					callback_frame = frame;
					sz = frame->actual_bytes;
					if (mFrameCallbackFunc) {
						callback_frame = temp;
						sz = callbackPixelBytes;
						int b = mFrameCallbackFunc(frame, temp);
						if (UNLIKELY(b)) {
							LOGW("failed to convert to callback frame");
							goto SKIP;
						}
					}
					jmethodID callbackOnFrameTemp = iframecallback_fields.onFrame;
					if (callbackOnFrameTemp) {
                                            jobject buf = env->NewDirectByteBuffer(callback_frame->data, callbackPixelBytes);
                                           env->CallVoidMethod(mFrameCallbackObj, iframecallback_fields.onFrame, buf);
                                           env->ExceptionClear();
                                           env->DeleteLocalRef(buf);
					}
				}
...
```