saki4510t/UVCCamera #251: On my project, it does not detect camera

irepan, 2017-10-13:
Hi, 

I need to create a camera service as usbCameraTest4, usbCameraTest4 works fine for me on my device, I have tried several ways, and as I do not have success, i just cloned usbCamera4 on a different project, but it looks like it is not detecting camera, I have cloned all configurations and copied all resources and main folders, even copied the proguard-project.txt, and it is not working, the weird thing is that usbCameraTest4 works fine, my project looks exactly the same but when I start OTG my project does not work and usbCameraTest4 works just fine, can somebody tell me wht do I need to do or to add to my project to make it work?

irepan, 2017-10-13:
Nevermind, I found the problem

closed 2017-10-13 by irepan

kolyadruz, 2020-11-20:
> Nevermind, I found the problem

Hi! What the problem is it? I have the same issue