saki4510t/UVCCamera #256: Getting black screen with Samsung Galaxy A and J series

dioi2000, 2017-10-25:
Hi Saki,

thank you for your project

its really work well but I have a problem that getting black screen with some device
(Galaxy A and J series and some of LG phones such as Nexus5X)

camera connected well but It's getting only black screen
I'm not sure what is the problem but it can't initialize setPreviewSize and I don't know how to fix it

saki4510t I need your help
thanks.

jp1017, 2017-10-26:
logs?

dioi2000, 2017-10-31:
@jp1017 here is a log

10-31 15:54:09.408 2804-5642/? W/System.err: java.lang.IllegalArgumentException: Failed to set preview size
10-31 15:54:09.413 2804-5642/? W/System.err:     at com.serenegiant.usb.UVCCamera.setPreviewSize(UVCCamera.java:328)
10-31 15:54:09.413 2804-5642/? W/System.err:     at com.serenegiant.usb.UVCCamera.setPreviewSize(UVCCamera.java:299)
10-31 15:54:09.414 2804-5642/? W/System.err:     at com.unichal.bookcam.service.UVCManager$1.onConnect(UVCManager.java:197)
10-31 15:54:09.414 2804-5642/? W/System.err:     at com.serenegiant.usb.USBMonitor$3.run(USBMonitor.java:574)
10-31 15:54:09.414 2804-5642/? W/System.err:     at android.os.Handler.handleCallback(Handler.java:789)
10-31 15:54:09.414 2804-5642/? W/System.err:     at android.os.Handler.dispatchMessage(Handler.java:98)
10-31 15:54:09.414 2804-5642/? W/System.err:     at android.os.Looper.loop(Looper.java:164)
10-31 15:54:09.414 2804-5642/? W/System.err:     at android.os.HandlerThread.run(HandlerThread.java:65)

++ sample 1,4 work well but others shutdown with error

jp1017, 2017-10-31:
> Failed to set preview size

dioi2000, 2017-11-01:
@jp1017 
I Know but I wanna know why failed and how to fix it

jp1017, 2017-11-01:
You may get the supported size in logs. 

So, What's the preview size you set?

dioi2000, 2017-11-01:
here is a supported size and preview size I set

supportedSize:{"formats":[{"index":1,"type":4,"default":1,"size":["640x480","352x288","320x240","176x144","160x120"]}]}

UVCCamera.DEFAULT_PREVIEW_WIDTH, UVCCamera.DEFAULT_PREVIEW_HEIGHT

dioi2000, 2017-11-06:
our source used old version of library

closed 2017-11-06 by dioi2000