saki4510t/UVCCamera #257: Can't connect to eachine ROTG01

njs236, 2017-10-26:
The two phones I have tested have had issues connecting to the eachine ROTG01.

Here's the stack trace.

https://pastebin.com/ezjdQVvj

https://pastebin.com/B6jzbr9Z

how do you switch logs on? can you access logs on the device? Im guessing that would help a bit more but it seems both are talking to libuvc at this point in the code so I went looking there for what was happening.

cephuo, 2017-11-15:
Any progress with this device?

njs236, 2017-11-19:
No sorry. We have a workable solution, so it wasn't urgent. but as far as using the program we are trying to build - we were looking for extra functions like taking photos or recording videos, which this offers.  It's okay for now if the issue was paused. 

I'll report on any progress when I get back to working on this.