saki4510t/UVCCamera #258: Crashes on runtime when used along JavaCV

yordanN, 2017-10-30:
When I use either JavaCV or UVCCamera alone, everything works fine. However, when I try to use both libraries in my project, JavaCV library files seem to disappear and I get the following error:

FATAL EXCEPTION: main
                                                                           Process: tv.mpact.genderagepredict, PID: 21313
                                                                           java.lang.UnsatisfiedLinkError: dalvik.system.PathClassLoader[DexPathList[[zip file "/data/app/tv.mpact.genderagepredict-2/base.apk", zip file "/data/app/tv.mpact.genderagepredict-2/split_lib_dependencies_apk.apk", zip file "/data/app/tv.mpact.genderagepredict-2/split_lib_slice_0_apk.apk", zip file "/data/app/tv.mpact.genderagepredict-2/split_lib_slice_1_apk.apk", zip file "/data/app/tv.mpact.genderagepredict-2/split_lib_slice_2_apk.apk", zip file "/data/app/tv.mpact.genderagepredict-2/split_lib_slice_3_apk.apk", zip file "/data/app/tv.mpact.genderagepredict-2/split_lib_slice_4_apk.apk", zip file "/data/app/tv.mpact.genderagepredict-2/split_lib_slice_5_apk.apk", zip file "/data/app/tv.mpact.genderagepredict-2/split_lib_slice_6_apk.apk", zip file "/data/app/tv.mpact.genderagepredict-2/split_lib_slice_7_apk.apk", zip file "/data/app/tv.mpact.genderagepredict-2/split_lib_slice_8_apk.apk", zip file "/data/app/tv.mpact.genderagepredict-2/split_lib_slice_9_apk.apk"],nativeLibraryDirectories=[/data/app/tv.mpact.genderagepredict-2/lib/arm, /data/app/tv.mpact.genderagepredict-2/base.apk!/lib/armeabi-v7a, /data/app/tv.mpact.genderagepredict-2/split_lib_dependencies_apk.apk!/lib/armeabi-v7a, /data/app/tv.mpact.genderagepredict-2/split_lib_slice_0_apk.apk!/lib/armeabi-v7a, /data/app/tv.mpact.genderagepredict-2/split_lib_slice_1_apk.apk!/lib/armeabi-v7a, /data/app/tv.mpact.genderagepredict-2/split_lib_slice_2_apk.apk!/lib/armeabi-v7a, /data/app/tv.mpact.genderagepredict-2/split_lib_slice_3_apk.apk!/lib/armeabi-v7a, /data/app/tv.mpact.genderagepredict-2/split_lib_slice_4_apk.apk!/lib/armeabi-v7a, /data/app/tv.mpact.genderagepredict-2/split_lib_slice_5_apk.apk!/lib/armeabi-v7a, /data/app/tv.mpact.genderagepredict-2/split_lib_slice_6_apk.apk!/lib/armeabi-v7a, /data/app/tv.mpact.genderagepredict-2/split_lib_slice_7_apk.apk!/lib/armeabi-v7a, /data/app/tv.mpact.genderagepredict-2/split_lib_slice_8_apk.apk!/lib/armeabi-v7a, /data/app/tv.mpact.genderagepredict-2/split_lib_slice_9_apk.apk!/lib/armeabi-v7a, /vendor/lib, /system/lib]]] couldn't find "libjniopencv_core.so"
                                                                               at java.lang.Runtime.loadLibrary(Runtime.java:367)
                                                                               at java.lang.System.loadLibrary(System.java:1076)
                                                                               at org.bytedeco.javacpp.Loader.loadLibrary(Loader.java:976)
                                                                               at org.bytedeco.javacpp.Loader.load(Loader.java:777)
                                                                               at org.bytedeco.javacpp.Loader.load(Loader.java:684)
                                                                               at org.bytedeco.javacpp.opencv_core.<clinit>(opencv_core.java:10)
                                                                               at java.lang.Class.classForName(Native Method)
                                                                               at java.lang.Class.forName(Class.java:324)
                                                                               at org.bytedeco.javacpp.Loader.load(Loader.java:739)
                                                                               at org.bytedeco.javacpp.Loader.load(Loader.java:684)
                                                                               at org.bytedeco.javacpp.opencv_objdetect$CascadeClassifier.<clinit>(opencv_objdetect.java:585)
                                                                               at tv.mpact.genderagepredict.Detection.FaceDetector.<init>(FaceDetector.java:18)
                                                                               at tv.mpact.genderagepredict.MainActivity.onCreate(MainActivity.java:85)
                                                                               at android.app.Activity.performCreate(Activity.java:6904)
                                                                               at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1136)
                                                                               at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3266)
                                                                               at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3415)
                                                                               at android.app.ActivityThread.access$1100(ActivityThread.java:229)
                                                                               at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1821)
                                                                               at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                               at android.os.Looper.loop(Looper.java:148)
                                                                               at android.app.ActivityThread.main(ActivityThread.java:7325)
                                                                               at java.lang.reflect.Method.invoke(Native Method)
                                                                               at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1230)
                                                                               at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1120)
                                                                            Caused by: java.lang.UnsatisfiedLinkError: dalvik.system.PathClassLoader[DexPathList[[zip file "/data/app/tv.mpact.genderagepredict-2/base.apk", zip file "/data/app/tv.mpact.genderagepredict-2/split_lib_dependencies_apk.apk", zip file "/data/app/tv.mpact.genderagepredict-2/split_lib_slice_0_apk.apk", zip file "

Thanks in advance.

yordanN, 2017-10-31:
The issue was that by including both libraries just like I normally do in my project, I ran into the java.lang.UnsatisfiedLinkError error. I ended up finding the following fix which allowed me to use both at the same time:

packagingOptions { exclude 'lib/armeabi-v7a/*.so' }

closed 2017-10-31 by yordanN

hasaxena, 2020-09-02:
> The issue was that by including both libraries just like I normally do in my project, I ran into the java.lang.UnsatisfiedLinkError error. I ended up finding the following fix which allowed me to use both at the same time:
> 
> packagingOptions { exclude 'lib/armeabi-v7a/*.so' }

This solution works in case of using object-tracker from Open CV in some other library or application.