saki4510t/UVCCamera #259: Native Memory leak

xiaozhi003, 2017-11-02:
Repeat it again and again.Native memory keeps growing.Android Studio3.0 can detect that.
@saki4510t Please help me,ths.
[image]

xiaozhi003, 2017-11-03:
OK，I found this bug
**UVCPreview.cpp**
```C
void UVCPreview::addCaptureFrame(uvc_frame_t *frame) {
	pthread_mutex_lock(&capture_mutex);
	if (LIKELY(isRunning())) {
		// keep only latest one
		if (captureQueu) {
			recycle_frame(captureQueu);
		}
		captureQueu = frame;
		pthread_cond_broadcast(&capture_sync);
	} else {
	    // Add this can sovle native leak 
	    recycle_frame(frame);
	}

	pthread_mutex_unlock(&capture_mutex);
}
```

closed 2017-11-03 by xiaozhi003

qiqisteve, 2018-04-19:
好像这个不起作用，还是有内存泄漏，及时不关闭都有

BarryAlertPan, 2018-05-07:
I also encountered this problem, how do you solve it?

xiaozhi003, 2018-05-08:
@BarryAlertPan According to the above modified source code **UVCPreview.cpp**

xiaozhi003, 2018-05-08:
@qiqisteve  这里肯定是有问题的，我改了这里循环开关24小时没有出现问题过，内存涨不过1-2M。还有一个linux文件句柄不能打开太多次（具体多少忘了好像是900次）否则报too many open files，所以**UsbControlBlock**也是可以复用的，不用每次都生成一个新的对象

myHelloer, 2019-07-29:
@xiaozhi003 你好，想问下关于UsbControlBlock的复用是怎么处理的啊

alless1, 2019-09-29:
It really doesn't work. My native heap keeps rising and can't be released.
Please help me。

YiZhou213, 2019-10-07:
After Modify 
---void UVCPreview::do_capture_callback(JNIEnv *env, uvc_frame_t *frame) {
ENTER();if (LIKELY(frame)) {
	uvc_frame_t *callback_frame = frame;
	if (mFrameCallbackObj) {
		if (mFrameCallbackFunc) {
			callback_frame = get_frame(callbackPixelBytes);
			if (LIKELY(callback_frame)) {
				int b = mFrameCallbackFunc(frame, callback_frame);
				recycle_frame(frame);
				if (UNLIKELY(b)) {
					LOGW("failed to convert for callback frame");
					goto SKIP;
				}
			} else {
				LOGW("failed to allocate for callback frame");
				callback_frame = frame;
				goto SKIP;
			}
		}
		jobject buf = env->NewDirectByteBuffer(callback_frame->data, callbackPixelBytes);
		pthread_mutex_lock(&capture_mutex);
		if(iframecallback_fields.onFrame != NULL){
			env->CallVoidMethod(mFrameCallbackObj, iframecallback_fields.onFrame, buf);
		}
		pthread_mutex_unlock(&capture_mutex);
		env->ExceptionClear();
		env->DeleteLocalRef(buf);
	}

And
void UVCPreview::addCaptureFrame(uvc_frame_t *frame) {
		}
		captureQueu = frame;
		pthread_cond_broadcast(&capture_sync);
	} else {
	    // Add this can sovle native leak
	    recycle_frame(frame);
	}

	pthread_mutex_unlock(&capture_mutex);
}
 
My native heap keeps rising too   and can't be released.

hugoleong, 2019-11-15:
感谢，解决了我遇到的问题！！！