saki4510t/UVCCamera #264: Compile Error?

panuniverse, 2017-11-12:
* What went wrong:
A problem occurred configuring project ':libuvccamera'.
> Could not resolve all dependencies for configuration ':libuvccamera:_debugPublishCopy'.
   > Could not resolve com.serenegiant:common:1.5.20.
     Required by:
         project :libuvccamera
      > Could not resolve com.serenegiant:common:1.5.20.
         > Could not get resource 'http://raw.github.com/saki4510t/libcommon/master/repository/com/serenegiant/common/1.5.20/common-1.5.20.pom'.
            > Could not GET 'http://raw.github.com/saki4510t/libcommon/master/repository/com/serenegiant/common/1.5.20/common-1.5.20.pom'.
               > raw.github.com

caseng, 2017-12-08:
I have the same problem. Is there a solution available?

caseng, 2017-12-15:
I found the solution for this problem. The problem was the firewall which did not trust a required certificate.  After disabling the firewall the project was build successfully. It may help as well building projects with WLAN  instead of using a LAN connection. The WLAN properties do not check certificates that strictly in some cases than LAN connections.