saki4510t/UVCCamera #265: [22820*UVCPreview.cpp:507:prepare_preview]:could not negotiate with camera:err=-51

ERICLIKESMILE, 2017-11-15:
when i used the Camera connect the Android phone, the log hint [22820*UVCPreview.cpp:507:prepare_preview]:could not negotiate with camera:err=-51,
I don't know the error code -51 what mean, and i have not found the document.
I need your help.
I am looking forward to your reply. thank you.

Tetralinks, 2017-12-21:
I got this error too. I solved this by change the third parameter of 

uvcCamera.setPreviewSize(width,height,
                UVCCamera.FRAME_FORMAT_MJPEG)
Maybe you can try.

WestbrookZero, 2019-05-17:
I also have the same problem，I change the third parameter of
uvcCamera.setPreviewSize(width,height,
UVCCamera.FRAME_FORMAT_MJPEG) or UVCCamera.FRAME_FORMAT_YUYV 

still cann't solve the probelm , I used 640x260 resolution.

rgm-22, 2020-10-20:
Same problem :(