saki4510t/UVCCamera #267: 镜像问题，前置摄像头（摄像头方向和屏幕方向一致）人左移视频中右移，人右移视频中左移，麻烦问下这个怎么解决

freedy-gb, 2017-11-20:


lkjustgo, 2017-11-27:
说英文啦

wangkang6, 2018-05-09:
Matrix matrix = new Matrix();
        DisplayMetrics dm = getResources().getDisplayMetrics();
        int w_screen = dm.widthPixels;
        int h_screen = dm.heightPixels;
        matrix.postScale(-1, 1, w_screen/2, h_screen / 2);//镜像水平翻转
//        matrix.postScale(1, -1, w_screen / 2, h_screen / 2);//镜像垂直翻转
        surfaceView.setTransform(matrix);

kerojiang, 2018-11-09:
> ```
>     Matrix matrix = new Matrix();
>     DisplayMetrics dm = getResources().getDisplayMetrics();
>     int w_screen = dm.widthPixels;
>     int h_screen = dm.heightPixels;
>     matrix.postScale(-1, 1, [...]

这个UVCCamera 直接把view传进去，拿不到原始的视频数据  怎么处理

kerojiang, 2018-11-09:
> _No description provided._

这个问题，后面解决了吗？

shirenhu, 2020-01-13:
我试过了，这样可行：
<com.serenegiant.widget.UVCCameraTextureView
		android:id="@+id/camera_view"
		android:rotation="90"
		android:scaleX="1.32"
		android:scaleY="1.32" ....

shengjunhu, 2020-07-13:
mabe you can add method for rotate mjpeg on UVCPreview.cpp