saki4510t/UVCCamera #273: Dealing with Cameras that overestimate their bandwidth Requirements

dxdy-name, 2017-11-23:
Hey there, 

I'm having trouble with two cameras that refuse to work together. I'm asking of them 640x480x30 fps video over the cellphones USB2.0 port. As I understand it even uncompressed that should be somewhat possible, but I'm using MJPEG frames, and (as each of the cameras works alone) realistically they range between 30kb and 90kb of image data worst case. So there should not be any bandwidth concerns.

In actuality however I cannot get them to work together, as every time I open the second camera (the order of which is which does not matter), this error occurs:

`E libuvc/stream: [19398*path\src\stream.c:1619:uvc_stream_start_bandwidth]:fail`

The function exists with an I/O error. I'm not using the full stack here, but rather directly use the modified libusb/libuvc from native code, which so far has not posed a problem, and different other cameras work fine together.

I believe this might be due to the cameras misdeclaring the bandwidth they need. Specifically, those are small embedded cameras, which makes this even more likely. The relevant descriptors are at the end of this post.

Moreover those cameras both have the same vendor id, and the same product id, and they export the exact same interface. Can this be a problem?

The fix-up block for cameras that fail to set dwMax is triggered as part of the startup.

While I am using an older phone (Galaxy S3), it can handle other cameras at other combinations that equal to a much higher actual bandwidth used without problems, so I do not believe the cellphone is at blame here.

# Attempted solutions

I have already attempted to mitigate the problem by manually setting dwMaxVideoFrameSize = 90000 and dwMaxPayloadTransferSize = 1024 manually in uvc_query_stream_ctrl, but that did not help with the problem. With these modifications the first camera still works perfectly fine, but the second one refuses to work at all.

It chooses an endpoint with endpoint_bytes_per_packet no matter how I set parameters, as that is the only endpoint the camera offers.

I have also tried playing around with the bandwidth parameter, but that did not improve the situation either.

# USB Interface (relevant sections)
`
     VideoStreaming Interface Descriptor:
        bLength                            34
        bDescriptorType                    36
        bDescriptorSubtype                  7 (FRAME_MJPEG)
        bFrameIndex                         3
        bmCapabilities                   0x01
          Still image supported
        wWidth                            640
        wHeight                           400
        dwMinBitRate                368640000
        dwMaxBitRate                737280000
        dwMaxVideoFrameBufferSize      768000
        dwDefaultFrameInterval          83333
        bFrameIntervalType                  2
        dwFrameInterval( 0)             83333
        dwFrameInterval( 1)            166666
      VideoStreaming Interface Descriptor:
        bLength                            30
        bDescriptorType                    36
        bDescriptorSubtype                  7 (FRAME_MJPEG)
        bFrameIndex                         4
        bmCapabilities                   0x01
          Still image supported
        wWidth                            640
        wHeight                           480
        dwMinBitRate                221184000
        dwMaxBitRate                221184000
        dwMaxVideoFrameBufferSize      921600
        dwDefaultFrameInterval         333333
        bFrameIntervalType                  1
        dwFrameInterval( 0)            333333
`

The delcarations of max video frame buffer size of 921600 and max bitrate in the 200Mbit range are what lead me to believe that they vastly overestimate the data for compressed jpeg (or even raw footage for that matter).

**How can I mitigate this problem so that I can use both of these cameras over the same host controller?**

closed 2017-11-24 by dxdy-name

FanHongchuang, 2018-02-07:
Hello, are you resolve it?

FanHongchuang, 2018-02-07:
@dxdy-name Hello, are you resolve it?

dxdy-name, 2018-02-07:
@FanHongchuang Kind of. There ar two cases that need to be distinguished, depending on the camera (and these issues are much more frequent with small embedded cameras). 

There is one case where the camera itself offers only a single transfer endpoint for 3x1024. In that case there's nothing you can do (at least to my knowledge) to get two of them working on a single endpoint. You can combine such a camera with another camera that barely uses any bandwidth and is very good (mostly involving the tricks below) but you're definitely wasting bandwidth.

In the case that the camera has many transfer endpoints but really the only problem is that it overestimates its bandwidth requirements, you can fix that. If you're on linux, you can have a look at the UVC_QUIRK_FIX_BANDWIDTH, if you're anywhere else you have to change that in your driver implementation. I'm not using the pristine libuvc from this repository, so I'm not sure how it's implemented here, but either the libuvc offers you a parameter (bandwidth?) that determines as a factor how much it's changed, or you have to get your hands dirty and modify the uvc_stream_start_bandwidth function (or whatever function you're using to initialize the stream) in stream.c so that it chooses an alternate endpoint and a numer of transfers that corresponds to what you deem an appropriate bandwith (this is something you may want to test, but I've encountered cameras where you can easily and with no errors get away with only using 1/10th of what they declare as the max bandwidth because they (somewhat justly) just put in the uncompressed data size as worst case for mjpeg-frames. But you definitely have to test that depending on your application with the camera.

Generally, if you want to gain insight into what's happening and which parameters are chosen that's the function to debug.

It's noteworthy that there is also a third forensic case. The code in the libuvc dependency here (at least at the version of a couple of months ago) explicitly assumes that the device enumerates its endpoints from low to high. There are however a few cameras that for whatever reason do not follow this convention, so what happens then is that even if they declare the correct bandwidth it goes through the alternate endpoints and encounters the highest one, but thinks that's the lowest that's fitting and choosing that. If that is the case (you can just check this with 'lsusb -d deviceidInHex:productIdInHex -v' connected to the linux dist of your choice) you also have to modify the stream.c file to make sure that it goes through all of them before choosing one. But that's a real edge case, I've so far only seen one camera that does that.

MelvinFMQ, 2019-04-05:
Am interested in this aswell. @dxdy-name

dxdy-name, 2019-04-05:
The issue had become moot, as the hardware in question did not export a suitable alternative interface endpoint, and thus it was fundamentally impossible to remedy in software.

MelvinFMQ, 2019-04-06:
@dxdy-name How do you determine that the hardware will provide an alternative interface? I'm trying to use 2 c920s resulting in bandwidth fail. Using a c270 with a c920 works but its not ideal as the c920 is much better camera.