saki4510t/UVCCamera #274: get keypress event

mahdipo, 2017-11-25:
hi saki
thanks to you for grate library
my gl uvc cam has 2 button.
one for snapshut and second for zoom.
 how can i get that button press event ?
i tried onkeydown event and onkey listener but they cant trigre by the buttons 
please help me to solve it

hossein37, 2018-01-28:
same problem..
with setButtonCallback method just received snapshot key, but zoom key didn't received, so i can't zoom with physical key on my digital microscope. also setZoom method did't work.
please guide me.
thank you.

shashankjamdade, 2018-06-04:
did you find out the solution for zoom ?? how??

mahdipo, 2018-06-04:
Hi i did'nt found any way to recive that damn key event and i stoped to
work on that.i hope you find it.
Please if you find tell me back too.
Thanks

hossein37, 2018-06-05:
Hi,yes i solved the problem.
does your camera really zoom or just scale the screen?

My usb camera have the zoom key,but doesn't really zoom,just scale the
screen,
So things to do:
1- connect your camera to your pc and when get the respond with any webcam
software, then use the 'usb trace' software(for windows) , click the zoom
button and check what data transfer between camera device and pc.

2- with 'usb host' library for android you can send and receive data
between device and usb camera.

3- connect your camera to android device and get the Unique data for zoom
button and simply respond to that data.

Good luck.

Christina-ko, 2021-12-21:
> ust received snapshot ke

Hi hossein37, 

Good morning. I realised that you are able to get the snapshot button callback from your uvc camera. Can you guide me on this. I am new with UVC camera. Which platform are you using ? (Android /Linux / Windows)
Please guide me !!!