saki4510t/UVCCamera #275: could not open camera:err=-1

coffeeplanter, 2017-11-26:
Hi, guys!

While launching the test apps I'm getting error:
`E/UVCCamera: [11204*UVCCamera.cpp:172:connect]:could not open camera:err=-1`

`usbcameratest`, `usbcameratest0`, `usbcameratest2` are also crashing with the following message:
```
E/UVCCamera: [7771*UVCCamera.cpp:172:connect]:could not open camera:err=-1
E/AndroidRuntime: FATAL EXCEPTION: BaseActivity
Process: com.serenegiant.usbcameratest0, PID: 7753
java.lang.UnsupportedOperationException: open failed:result=-1
at com.serenegiant.usb.UVCCamera.open(UVCCamera.java:204)
at com.serenegiant.usbcameratest0.MainActivity$2$1.run(MainActivity.java:149)
at android.os.Handler.handleCallback(Handler.java:751)
at android.os.Handler.dispatchMessage(Handler.java:95)
at android.os.Looper.loop(Looper.java:154)
at android.os.HandlerThread.run(HandlerThread.java:61)
```

Using the latest sdk 27 and ndk 16, Logitech cam, and Samsung SM-A710F device.

Help me please to find the reason of the error and fix it.

grliao, 2017-11-27:
may be this can help you.
https://github.com/saki4510t/UVCCamera/issues/181

MichaelLeeJones, 2017-11-30:
You can try switching the NDK version to 12

dmitrykcom, 2017-11-30:
I'm having the same issue with Inogeni and Pixel on 8.0. Switching to NDK 14b didn't help.

zangpuu, 2017-12-21:
I also had same problems with NDK15 and eventually found that clang was the reason.
Then I change the compiler to gcc by uncomment "NDK_TOOLCHAIN_VERSION := 4.9" from "libuvccamera/src/main/jni/Application.mk".
And it works now.

shibbs, 2018-02-28:
this solved the issue for me, thank you @zangpuu

shibbs, 2018-03-22:
@dmitrykcom This fixed it for me on a different phone/camera setup but I'm now also having issues with the inogeni and a pixel 2. Did you have any success?

HuitMahoon, 2019-06-01:
> I also had same problems with NDK15 and eventually found that clang was the reason.
> Then I change the compiler to gcc by uncomment "NDK_TOOLCHAIN_VERSION := 4.9" from "libuvccamera/src/main/jni/Application.mk".
> And it works now.

this works for me ， thanks

zipswich, 2024-05-24:
After a series of recent Android SDK package updates, this error started to happen. I confirmed this by making sure the code is exactly the same as it was a few weeks ago when it was working.  There were so many updates that it is impossible to pinpoint the culprit update.