saki4510t/UVCCamera #277: ABI "armeabi" is deprecated in r16 and will be removed in r17.

timsheu, 2017-11-28:
Hi saki4510t, 

Thank you very much for providing such great project! It helps me a lot!
I faced this issue couple of days ago as upgrading my NDK from r15 to r16. The gradle build system says something as title. This limitation is listed on [official site](https://developer.android.com/ndk/guides/abis.html#sa).
I wonder if you have a plan on handling this? If not, would you provide some instruction to deal with?
Thanks again for the project!

Best regards, 
Tim.

Mr-Grieves, 2017-12-04:
I am currently stuck on this issue.

Is there any way around this, other than downgrading my NDK to r15?

Thanks,
Nathan

xbesung, 2017-12-15:
Edit UVCCamera/libuvccamera/src/main/jni/Application.mk, delete "armeabi" and "mips" (also deprecated) in APP_ABI.

timsheu, 2017-12-18:
Hi @xbesung, 
The test5 app crashes after I do the changes in libuvccamera library you mentioned. The NDK build process is success. Any ideas?