saki4510t/UVCCamera #282: It works in debug mode with Android studio. However, in release mode, the following error occurs.

world9604, 2017-12-07:
It works in debug mode with Android studio. However, in release mode, the following error occurs.

My UVCCamera library was created in 2014-2015.

And in release mode, the following error is dropped.

> 12-06 22:26:28.673 11423-11439/? E/art: Failed to register native method com.serenegiant.usb.UVCCamera.nativeSetStatusCallback(JLcom/serenegiant/usb/IStatusCallback;)I in /data/app/com.serenegiant.ubkeyirisuvc-1/base.apk
> 12-06 22:26:28.676 11423-11439/? E/art: ----- class 'Lcom/serenegiant/usb/UVCCamera;' cl=0x12c419a0 -----
> 12-06 22:26:28.677 11423-11439/? E/art:   objectSize=974 (432 from super)
> 12-06 22:26:28.677 11423-11439/? E/art:   access=0x0008.0001
> 12-06 22:26:28.677 11423-11439/? E/art:   super='java.lang.Class<java.lang.Object>' (cl=0x0)
> 12-06 22:26:28.677 11423-11439/? E/art:   vtable (72 entries, 11 in super):
> 12-06 22:26:28.677 11423-11439/? E/art:      0: boolean com.serenegiant.usb.UVCCamera.checkSupportFlag(long)
> 12-06 22:26:28.677 11423-11439/? E/art:      1: void com.serenegiant.usb.UVCCamera.close()
> 12-06 22:26:28.677 11423-11439/? E/art:      2: void com.serenegiant.usb.UVCCamera.destroy()
> 12-06 22:26:28.677 11423-11439/? E/art:      3: boolean com.serenegiant.usb.UVCCamera.getAutoFocus()
> 12-06 22:26:28.677 11423-11439/? E/art:      4: boolean com.serenegiant.usb.UVCCamera.getAutoWhiteBlance()
> 12-06 22:26:28.677 11423-11439/? E/art:      5: int com.serenegiant.usb.UVCCamera.getBrightness()
> 12-06 22:26:28.677 11423-11439/? E/art:      6: int com.serenegiant.usb.UVCCamera.getBrightness(int)
> 12-06 22:26:28.677 11423-11439/? E/art:      7: int com.serenegiant.usb.UVCCamera.getContrast()
> 12-06 22:26:28.677 11423-11439/? E/art:      8: int com.serenegiant.usb.UVCCamera.getContrast(int)
> 12-06 22:26:28.677 11423-11439/? E/art:      9: android.hardware.usb.UsbDevice com.serenegiant.usb.UVCCamera.getDevice()
> 12-06 22:26:28.677 11423-11439/? E/art:     10: java.lang.String com.serenegiant.usb.UVCCamera.getDeviceName()
> 12-06 22:26:28.677 11423-11439/? E/art:     11: int com.serenegiant.usb.UVCCamera.getFocus()
> 12-06 22:26:28.677 11423-11439/? E/art:     12: int com.serenegiant.usb.UVCCamera.getFocus(int)
> 12-06 22:26:28.677 11423-11439/? E/art:     13: int com.serenegiant.usb.UVCCamera.getGain()
> 12-06 22:26:28.677 11423-11439/? E/art:     14: int com.serenegiant.usb.UVCCamera.getGain(int)
> 12-06 22:26:28.677 11423-11439/? E/art:     15: int com.serenegiant.usb.UVCCamera.getGamma()
> 12-06 22:26:28.677 11423-11439/? E/art:     16: int com.serenegiant.usb.UVCCamera.getGamma(int)
> 12-06 22:26:28.677 11423-11439/? E/art:     17: int com.serenegiant.usb.UVCCamera.getHue()
> 12-06 22:26:28.677 11423-11439/? E/art:     18: int com.serenegiant.usb.UVCCamera.getHue(int)
> 12-06 22:26:28.677 11423-11439/? E/art:     19: int com.serenegiant.usb.UVCCamera.getPowerlineFrequency()
> 12-06 22:26:28.677 11423-11439/? E/art:     20: com.serenegiant.usb.f com.serenegiant.usb.UVCCamera.getPreviewSize()
> 12-06 22:26:28.677 11423-11439/? E/art:     21: int com.serenegiant.usb.UVCCamera.getSaturation()
> 12-06 22:26:28.677 11423-11439/? E/art:     22: int com.serenegiant.usb.UVCCamera.getSaturation(int)
> 12-06 22:26:28.677 11423-11439/? E/art:     23: int com.serenegiant.usb.UVCCamera.getSharpness()
> 12-06 22:26:28.677 11423-11439/? E/art:     24: int com.serenegiant.usb.UVCCamera.getSharpness(int)
> 12-06 22:26:28.677 11423-11439/? E/art:     25: java.lang.String com.serenegiant.usb.UVCCamera.getSupportedSize()
> 12-06 22:26:28.677 11423-11439/? E/art:     26: java.util.List com.serenegiant.usb.UVCCamera.getSupportedSizeList()
> 12-06 22:26:28.677 11423-11439/? E/art:     27: com.serenegiant.usb.g$b com.serenegiant.usb.UVCCamera.getUsbControlBlock()
> 12-06 22:26:28.677 11423-11439/? E/art:     28: int com.serenegiant.usb.UVCCamera.getWhiteBlance()
> 12-06 22:26:28.677 11423-11439/? E/art:     29: int com.serenegiant.usb.UVCCamera.getWhiteBlance(int)
> 12-06 22:26:28.677 11423-11439/? E/art:     30: int com.serenegiant.usb.UVCCamera.getZoom()
> 12-06 22:26:28.677 11423-11439/? E/art:     31: int com.serenegiant.usb.UVCCamera.getZoom(int)
> 12-06 22:26:28.677 11423-11439/? E/art:     32: void com.serenegiant.usb.UVCCamera.open(com.serenegiant.usb.g$b)
> 12-06 22:26:28.677 11423-11439/? E/art:     33: void com.serenegiant.usb.UVCCamera.resetBrightness()
> 12-06 22:26:28.677 11423-11439/? E/art:     34: void com.serenegiant.usb.UVCCamera.resetContrast()
> 12-06 22:26:28.677 11423-11439/? E/art:     35: void com.serenegiant.usb.UVCCamera.resetFocus()
> 12-06 22:26:28.677 11423-11439/? E/art:     36: void com.serenegiant.usb.UVCCamera.resetGain()
> 12-06 22:26:28.677 11423-11439/? E/art:     37: void com.serenegiant.usb.UVCCamera.resetGamma()
> 12-06 22:26:28.677 11423-11439/? E/art:     38: void com.serenegiant.usb.UVCCamera.resetHue()
> 12-06 22:26:28.677 11423-11439/? E/art:     39: void com.serenegiant.usb.UVCCamera.resetSaturation()
> 12-06 22:26:28.677 11423-11439/? E/art:     40: void com.serenegiant.usb.UVCCamera.resetSharpness()
> 12-06 22:26:28.677 11423-11439/? E/art:     41: void com.serenegiant.usb.UVCCamera.resetWhiteBlance()
> 12-06 22:26:28.677 11423-11439/? E/art:     42: void com.serenegiant.usb.UVCCamera.resetZoom()
> 12-06 22:26:28.677 11423-11439/? E/art:     43: void com.serenegiant.usb.UVCCamera.setAutoFocus(boolean)
> 12-06 22:26:28.677 11423-11439/? E/art:     44: void com.serenegiant.usb.UVCCamera.setAutoWhiteBlance(boolean)
> 12-06 22:26:28.677 11423-11439/? E/art:     45: void com.serenegiant.usb.UVCCamera.setBrightness(int)
> 12-06 22:26:28.677 11423-11439/? E/art:     46: void com.serenegiant.usb.UVCCamera.setButtonCallback(com.serenegiant.usb.c)
> 12-06 22:26:28.677 11423-11439/? E/art:     47: void com.serenegiant.usb.UVCCamera.setCamera(int, int, boolean, int)
> 12-06 22:26:28.677 11423-11439/? E/art:     48: void com.serenegiant.usb.UVCCamera.setContrast(int)
> 12-06 22:26:28.677 11423-11439/? E/art:     49: void com.serenegiant.usb.UVCCamera.setFocus(int)
> 12-06 22:26:28.677 11423-11439/? E/art:     50: void com.serenegiant.usb.UVCCamera.setFrameCallback(com.serenegiant.usb.d, int)
> 12-06 22:26:28.677 11423-11439/? E/art:     51: void com.serenegiant.usb.UVCCamera.setGain(int)
> 12-06 22:26:28.677 11423-11439/? E/art:     52: void com.serenegiant.usb.UVCCamera.setGamma(int)
> 12-06 22:26:28.677 11423-11439/? E/art:     53: void com.serenegiant.usb.UVCCamera.setHue(int)
> 12-06 22:26:28.677 11423-11439/? E/art:     54: void com.serenegiant.usb.UVCCamera.setPowerlineFrequency(int)
> 12-06 22:26:28.677 11423-11439/? E/art:     55: void com.serenegiant.usb.UVCCamera.setPreviewDisplay(android.view.Surface)
> 12-06 22:26:28.677 11423-11439/? E/art:     56: void com.serenegiant.usb.UVCCamera.setPreviewDisplay(android.view.SurfaceHolder)
> 12-06 22:26:28.677 11423-11439/? E/art:     57: void com.serenegiant.usb.UVCCamera.setPreviewSize(int, int)
[... 329 lines cut ...]
> 12-06 22:26:28.681 11423-11439/? E/art:     110: int com.serenegiant.usb.UVCCamera.mWhiteBlanceRelDef
> 12-06 22:26:28.681 11423-11439/? E/art:     111: int com.serenegiant.usb.UVCCamera.mWhiteBlanceRelMax
> 12-06 22:26:28.681 11423-11439/? E/art:     112: int com.serenegiant.usb.UVCCamera.mWhiteBlanceRelMin
> 12-06 22:26:28.681 11423-11439/? E/art:     113: int com.serenegiant.usb.UVCCamera.mZoomDef
> 12-06 22:26:28.681 11423-11439/? E/art:     114: int com.serenegiant.usb.UVCCamera.mZoomMax
> 12-06 22:26:28.681 11423-11439/? E/art:     115: int com.serenegiant.usb.UVCCamera.mZoomMin
> 12-06 22:26:28.681 11423-11439/? E/art:     116: int com.serenegiant.usb.UVCCamera.mZoomRelDef
> 12-06 22:26:28.681 11423-11439/? E/art:     117: int com.serenegiant.usb.UVCCamera.mZoomRelMax
> 12-06 22:26:28.681 11423-11439/? E/art:     118: int com.serenegiant.usb.UVCCamera.mZoomRelMin
> 12-06 22:26:28.681 11423-11439/? E/libUVCCamera: [11439*serenegiant_usb_UVCCamera.cpp:1984:registerNativeMethods]:registerNativeMethods failed(class=com/serenegiant/usb/UVCCamera)
>                                                  
>                                                  --------- beginning of crash
> 12-06 22:26:28.682 11423-11439/? E/AndroidRuntime: FATAL EXCEPTION: CameraThread
>                                                    Process: com.serenegiant.ubkeyirisuvc, PID: 11423
[... cut ...]
>                                                        at com.serenegiant.ubkeyirisuvc.MainActivity$a$a.a(Unknown Source)
>                                                        at com.serenegiant.ubkeyirisuvc.MainActivity$a.handleMessage(Unknown Source)
>                                                        at android.os.Handler.dispatchMessage(Handler.java:102)
>                                                        at android.os.Looper.loop(Looper.java:154)
>                                                        at com.serenegiant.ubkeyirisuvc.MainActivity$a$a.run(Unknown Source)

world9604, 2017-12-07:
Sorry for the error log being too long.
Please copy only the error log part and paste it on the notepad.

Sean-Kone, 2018-01-31:
I also got this error: java.lang.NoSuchMethodError: no static or non-static method "Lcom/serenegiant/usb/UVCCamera;.nativeSetStatusCallback(JLcom/serenegiant/usb/IStatusCallback;)I"

The issue for me was with pro guard.
I resolved it by adding
-keep class com.serenegiant.** { *; }

to the pro guard rules file.

hope that helps

MADHUSUDHANRAO527, 2018-05-10:
thanks @Sean-Kone Its working

shashankjamdade, 2018-06-04:
I am developing the android app having external USB-camera functinality. I want to integrate zoom functionality but 'setZoom(final int zoom)' function which is available in library is not working properly. I am setting different int values & I tried to print the current zoom value it always return me -11 . I'm not able to see any changes on zoom in my camera view. please help me as soon as possible .

shashankjamdade, 2018-06-26:
https://github.com/saki4510t/UVCCamera/issues/367

ChineseBoyLY, 2019-06-27:
buildTypes {
        release {
            minifyEnabled false //don`t
            proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-project.txt'
        }
    }

heaveninfosoft, 2020-06-24:
> buildTypes {
> release {
> minifyEnabled false //don`t
> proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-project.txt'
> }
> }

hello how i can i build app for dual usb camera can support hd and fhd can  you help me???

mikite, 2021-11-24:
In my case, it crashes  when I tried to call :
**`mNativePtr = nativeCreate();`**
pls help me to figure out what's the reason,  thanks a lot!
Here is the output logs:
 java.lang.UnsatisfiedLinkError: No implementation found for long com.zzw.mointor.UVCCamera.nativeCreate() (tried Java_com_ xxx_UVCCamera_nativeCreate and Java_com_zzw_mointor_UVCCamera_nativeCreate__)