saki4510t/UVCCamera #286: How to use UVCCamera with jitpack?

pe78420, 2017-12-14:
Hi,

Is there a way to to get your UVCCamera project into my own build using jitpack?

Like this:

In root build.gradle:
allprojects { 
    repositories { 
          ...
          maven { url 'https://jitpack.io' }
    }
}
In app build.gradle:
dependencies {
     implementation 'com.github.saki4510t:UVCCamera:?????'
}

The challenge is to replace the "????" with the right version number :(

Unfortunately, I got systematicaly the following error:
Could not find com.github.saki4510t:UVCCamera:?????

What's wrong?

Many thanks in advance and best regards...

Philippe

jp1017, 2017-12-15:
You should build this as an aar lib.

pe78420, 2017-12-15:
Thanks for your answer... but I'm a newbie in Android dev ;)
How to do that ?
Do I have to import your project into Android Studio first and then build it as an aar ?

jp1017, 2017-12-15:
Yes, 
1. import this to Android Studio
1. compile this, you may get aar in /build/outputs/aar/
1. add aar to you app's lib dir.
1. edit build.gradle: compile(name: 'libuvccamera-release', ext: 'aar')

pe78420, 2017-12-15:
OK thanks, I will try it... but in this case, it seems that jitpack is not used since your lib is already downloaded!

pe78420, 2017-12-15:
Which librairies I have to download? only libuvccamera? anything else?

pe78420, 2017-12-17:
Hi,

Finaly I tried using Bintray Jcenter and that's work ;)

But I've some troubles to understand the logical of your API...
Do you have any sort of diagram which explain which API have to use in first and so on?
USBMonitor, UVCCameraHandler, CameraViewInterface, UVCCameraTextureView, etc.

Which ones are mandatory?
Which ones are optional?

I've some difficulties to find the meaning of these differents classes inside the java code :(

Thanks in advance and best regards...

Philippe

closed 2017-12-17 by pe78420