saki4510t/UVCCamera #288: How can change the camera preview size dynamically?

rkdalsrb, 2017-12-15:
I want to change the camera preview size, for example between 640x480 and 320x240, dynamically  in the middle of program. Only mUVCCamera.setPreviewSize method does not work.  
Please give me some help. Thanks in advance.

mayankjain95, 2018-05-08:
Hi
Yes, you can change the camera preview size dynamically. First, make mCameraHandler null, pause the mUVCCameraView. Then create new mCameraHandler with the new preview size and change the aspect ratio using mUVCCameraView.setAspectRatio.
Then resume the mUVCCameraVIew and start the preview.
Hope it helps.

rgm-22, 2021-09-14:
Same issue :(

Did you solve it @rkdalsrb?

xuannt88, 2023-05-11:
> Hi Yes, you can change the camera preview size dynamically. First, make mCameraHandler null, pause the mUVCCameraView. Then create new mCameraHandler with the new preview size and change the aspect [...]

Could you note more details (ex: sample code)