saki4510t/UVCCamera #294: The camera preview not working

gallavie, 2017-12-20:
Hello everyone
The problem:
The project is running but the camera preview not working

1) I downloaded the repo 
2) I changed sdk and ndk path like this
sdk.dir=myPath/android-sdks
ndk.dir=myPath/android-ndk-r14b
(I used ndk 14 and 16 because i saw there is a problem with ndk 15 https://github.com/saki4510t/UVCCamera/issues/181)
3) I run the project and click on  buttonto turn on preview camera.
the device detect the camera but the preview not working 
4) If i run application like MScope or UsbWebCamera so the camera preview working 
5) I added screenshots of my application

[image]
[image]

what the problem?
thank you

brunnodev, 2018-03-12:
Same problem here 😞 

Did you solve the problem?

brunnodev, 2018-03-19:
I figured out that the problem was on my cell phone.

I try in another and work like a charm 👍 👍 

🎉