saki4510t/UVCCamera #296: UnsupportedOperationException: open failed:result=-1

apples398, 2017-12-21:
Process: com.serenegiant.usbcameratest, PID: 5575
                                                 java.lang.UnsupportedOperationException: open failed:result=-1
                                                     at com.serenegiant.usb.UVCCamera.open(UVCCamera.java:204)
                                                     at com.serenegiant.usbcameratest.MainActivity$2$1.run(MainActivity.java:142)
                                                     at android.os.Handler.handleCallback(Handler.java:739)
                                                     at android.os.Handler.dispatchMessage(Handler.java:95)
                                                     at android.os.Looper.loop(Looper.java:135)
                                                     at android.os.HandlerThread.run(HandlerThread.java:61)

apples398, 2017-12-22:
Process: com.serenegiant.usbcameratest, PID: 1736
                                                 java.lang.UnsupportedOperationException: open failed:result=-50
                                                     at com.serenegiant.usb.UVCCamera.open(UVCCamera.java:204)

lohithn4, 2017-12-27:
me too facing same issue

lohithn4, 2017-12-27:
[https://github.com/saki4510t/UVCCamera/issues/275](url)
fixed the issue for me

closed 2018-01-01 by apples398

yeshiyuan00, 2018-04-14:
page not found 404

CreazyBlade, 2018-04-23:
page not found 404

RaimuYeh, 2019-02-14:
> page not found 404

The link is not correctly