saki4510t/UVCCamera #297: UVC Camera Xperia Go (Novathor Mali400)

Peter-St, 2017-12-23:
Hello,

I'm trying to get an uvc camera on a sony xperia go smartphone (novathor with mali400 gpu) to run. I have already included the uvc in the Kernel and set the permissions of the /dev/video0 node to 777. The firmware of the smartphone is JellyBean 4.1.2. I have compiled the apps from your github (UVCCamera) and started the app USBCameraTest, but I didn't get the camera to work. There is a logcat and dmesg of the process attached.

Perhaps anyone

 can tell me, why I'm not gettin a UVC Camera on my St27i to work. I have already tryed other apps (USBCameraTest0, Dashcam, Simplewebcam, CameraFi) from the market, but without sucess. Please have a quick look at it.

Best Wishes,

Peter

[cat .proc.last_kmsg.txt](https://github.com/saki4510t/UVCCamera/files/1584295/cat.proc.last_kmsg.txt)
[LOGCAT.txt](https://github.com/saki4510t/UVCCamera/files/1584296/LOGCAT.txt)

timsheu, 2017-12-28:
[https://github.com/saki4510t/UVCCamera#20140725](https://github.com/saki4510t/UVCCamera#20140725)

jp1017, 2017-12-28:
You don't need to change the permission of /dev/video0 for this lib doesn't use v4l2.

You may change to another phone to test.

Peter-St, 2017-12-28:
ok,
the xperia go might have low usbhost compatibilties. Only thing would be to apply some patches to the kernelsource or libusb source. Perhaps someone have already successfully applied them,  then please let me know.

Regards,

Peter

-> Failure log attached

[logcat_uvcam.txt](https://github.com/saki4510t/UVCCamera/files/1591500/logcat_uvcam.txt)

Peter-St, 2018-02-04:
UPDATE:

By the while I was still able to get some Frames from my camera. The specific thing is, that the device only can submit frames, which were sent by low maxPacketSize. Most cameras support more than one maxPacketSizes to submit the stream and here is a difference in specific Chipsets of the Telephons of MTK and Novathor which only submits a low maxPacketSize. My qualcom devices receives all the maxPacketSizes the camera can send.
Some more Info about Uvc stream can be found here --> [http://www.usb.org/developers/docs/devclass_docs/](url)

Also it was easier for the devices to get the frames in MJPEG format that uncompressed YUV.

So far,

Peter

closed 2021-07-03 by Peter-St