saki4510t/UVCCamera #299: clarification for live debugging with  UVC camera and android-studio

lohithn4, 2017-12-27:
Hi, everyone,
This is one of the great project to work with.

I am able to run the samples on the android device. then i'm copying those apks into my mobile to run the apk.

I need one clarification, i don't know whether this is a valid question,
I just not able to understand how everyone are using this app for debugging.

Is it possible to connect UVC camera to mobile and use it at the same with android-studio to debug the app live!!!.

Why i am asking this question is,, i am using redmi-note-4 which has only one port for connecting the UVC-camera.

I am stuck at this point, can somebody point me with the right guidance to proceed forward to connect- mobile + UVC-camera+ android-studio to debug app in parallel. 

Even some general information about this app can be used to debug live by connecting to my lap-top is great.

Thanks.

jp1017, 2017-12-28:
You need the OTG line.

lohithn4, 2017-12-28:
Thanks, for your reply, just one more clarification, if I connect usbCamera + mobile <-- android_studio with this cable(attached photo) will it work for operating in parallel. 

From: 蒋朋
Sent: 28 December 2017 05:48
To: saki4510t/UVCCamera
Cc: Lohith; Author
Subject: Re: [saki4510t/UVCCamera] clarification for live debugging with UVCcamera and android-studio (#299)

You need the OTG line.

timsheu, 2017-12-28:
You can use adb through Wi-Fi If your device is rooted.
Additionally, open an text file and write the log.
Steps:
1. Write log like [this thread's answer](https://stackoverflow.com/questions/14709773/android-ndk-write-logs-to-a-file)
(I myself change the log path to "/data/data/com.myfolder.test/log.txt" because my device cannot be rooted, damned OP#O company)
2. In command line or terminal: adb shell run-as 'your-app-bundle-package-full-path'
(e.g. com.myfolder.test)
3. You should now see the log file.

Since my device disallows me to pull out the log, I use "cat log.txt" to debug...Pain in the ass but at least worked ;)

maciejgrabek, 2018-01-31:
Hi!
I'm using this way: https://github.com/saki4510t/UVCCamera/wiki/howto_debug_over_network - same solution was on stackoverflow - it is working nice. If android studio is not showing the device for debugging just type adb connect IP:5555 once again and it will work.