saki4510t/UVCCamera #300: i cant get it working on usbtv007 devices like this

Robokishan, 2017-12-30:
idVendor           0x1b71 Fushicai
  idProduct          0x1b71 USBTV007 Video Grabber [EasyCAP]

this vendor and product id please help 
many app on play store use your library to support their app but your own app does not support it please help

jp1017, 2018-01-02:
error logcat?

Robokishan, 2018-01-12:
Usb device is not showing in list

FeldmanItamar, 2018-03-08:
Hi, is there any solution here?

rami-dabain, 2018-05-01:
adding 
`<usb-device product-id="12290" vendor-id="7025"/>`
to the device_filter.xml makes the device show in the list
however i couldn't get mine to work

Robokishan, 2018-06-05:
listing is available on mine but I can't get it work because frame which is returned by the device is something else which is not available in the code we have to add it separately