saki4510t/UVCCamera #304: Unable to build project

ravikumarg45, 2018-01-11:
Hi , I am trying to build the project. but i get the following error:

Error:Could not resolve all files for configuration ':libuvccamera:debugCompileClasspath'.
> Could not resolve com.serenegiant:common:1.5.20.
  Required by:
      project :libuvccamera
   > Could not resolve com.serenegiant:common:1.5.20.
      > Could not get resource 'http://raw.github.com/saki4510t/libcommon/master/repository/com/serenegiant/common/1.5.20/common-1.5.20.pom'.
         > Could not GET 'http://raw.github.com/saki4510t/libcommon/master/repository/com/serenegiant/common/1.5.20/common-1.5.20.pom'. Received status code 403 from server: Forbidden
   > Could not resolve com.serenegiant:common:1.5.20.
      > Could not get resource 'https://jcenter.bintray.com/com/serenegiant/common/1.5.20/common-1.5.20.pom'.
         > Could not GET 'https://jcenter.bintray.com/com/serenegiant/common/1.5.20/common-1.5.20.pom'.
            > Connection reset

jp1017, 2018-01-12:
Try version 1.5.4

ravikumarg45, 2018-01-15:
Firewall issue, works fine.

closed 2018-01-15 by ravikumarg45