saki4510t/UVCCamera #309: Camera preview is not getting any tablet phone

mirzaaqib, 2018-01-31:
I have successfully getting the preview in all the devices (Smartphones).But the same application is not working with the Tablet device. Camera getting connected & turn on but the preview is not coming also i m not getting any error in log.Please help me out & provide me the solution if possible.

Thank You!!

jp1017, 2018-01-31:
which tablet? Your own device?

mirzaaqib, 2018-01-31:
I tried with Samsung, Hp, & iball & micromax some models.

binilj04, 2018-02-01:
Not working  on Samsung On7 and Samsung Galaxy Tab S2.

binilj04, 2018-02-03:
Is there a way to view debug logs? Since the camera is connected via usb.

maciejgrabek, 2018-02-04:
@binilj04 You can debug remotely. 
While connected via USP with your dev device:
adb shell ifconfig

adb tcpip 5555

adb connect <DEVICE_IP_ADDRESS>:5555 

then unplug - and you can use the device over WiFi. 

when you are done:
adb -s <DEVICE_IP_ADDRESS>:5555 usb
to turn WiFi debugging off (for security :))

check also this page: https://github.com/saki4510t/UVCCamera/wiki/howto_debug_over_network