saki4510t/UVCCamera #316: Con't find some file

lcxiaozi, 2018-02-07:
Dear friends

Some files con't find in the project:

import com.serenegiant.utils.BuildCheck;
import com.serenegiant.utils.HandlerThreadHandler;

where are they?

wly65535, 2018-02-07:
Add these line to your `build.gradle`

```
repositories {
        maven { url 'http://raw.github.com/saki4510t/libcommon/master/repository/' }
        jcenter()
    }

compile('com.serenegiant:common:1.5.20') {
        exclude module: 'support-v4'
    }
```

cihangoksu, 2021-11-13:
Hi I'm tring to implement your repo with kivy and buildozer, but get the similar error "error: cannot find symbol:  import com.serenegiant.utils.BuildCheck;".. I tried to modify the buildozer spec file to exclude 'support-v4', but it didn't help.. Is there any other way to solve this issue?? Thanks!