saki4510t/UVCCamera #321: could not open camera:err=-50

cisankit, 2018-02-19:
I did downgrade the ndk version to 14b. still getting the error while opening camera.

02-19 19:19:06.571 28018-28050/com.serenegiant.usbcameratest E/libuvc/device: [28050*device.c:944:uvc_scan_control]:UVC_ERROR_INVALID_DEVICE
02-19 19:19:06.571 28018-28050/com.serenegiant.usbcameratest E/UVCCamera: [28050*UVCCamera.cpp:172:connect]:could not open camera:err=-50
02-19 19:19:06.571 28018-28050/com.serenegiant.usbcameratest E/AndroidRuntime: FATAL EXCEPTION: BaseActivity
                                                                               Process: com.serenegiant.usbcameratest, PID: 28018
                                                                               java.lang.UnsupportedOperationException: open failed:result=-50
                                                                                   at com.serenegiant.usb.UVCCamera.open(UVCCamera.java:204)
                                                                                   at com.serenegiant.usbcameratest.MainActivity$2$1.run(MainActivity.java:142)
                                                                                   at android.os.Handler.handleCallback(Handler.java:739)
                                                                                   at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                                   at android.os.Looper.loop(Looper.java:145)
                                                                                   at android.os.HandlerThread.run(HandlerThread.java:61)

davidhowe, 2019-03-05:
@ankitm099 I'm also receiving this error on a Samsung S4. @saki4510t any comment on this?

huasheng0401, 2019-03-23:
I have the same issue:
e = {UnsupportedOperationException@4238} "java.lang.UnsupportedOperationException: open failed:result=-50----->id_camera=2941849728;venderId=3034;productId=46880;fileDescriptor=65;busNum=1;devAddr=5;usbfs=/dev/bus/usb\nException:nativeConnect return value: -50"
 cause = {UnsupportedOperationException@4238} "java.lang.UnsupportedOperationException: open failed:result=-50----

arpitcybernetyx, 2020-09-03:
Having same issues 
```
2020-09-03 17:28:41.756 29245-29366/com.jiangdg.usbcamera E/libuvc/device: [29366*device.c:944:uvc_scan_control]:UVC_ERROR_INVALID_DEVICE
2020-09-03 17:28:41.756 29245-29366/com.jiangdg.usbcamera E/UVCCamera: [29366*UVCCamera.cpp:172:connect]:could not open camera:err=-50
2020-09-03 17:30:18.999 29245-29366/com.jiangdg.usbcamera E/libuvc/device: [29366*device.c:944:uvc_scan_control]:UVC_ERROR_INVALID_DEVICE
2020-09-03 17:30:18.999 29245-29366/com.jiangdg.usbcamera E/UVCCamera: [29366*UVCCamera.cpp:172:connect]:could not open camera:err=-50
```

paijwar, 2020-09-03:
Facing the same issue with OnePlus 7 & ezCap UVC device, any help appreciated.

CarryHekangkai, 2020-09-11:
我也同样遇到了这个问题。我发现出现这个问题的设备，在git拉来的最新代码运行过程中会出现，但是在下载的apk中却能正常运行，希望作者能给出apk中的解决方案。

paijwar, 2020-09-11:
@CarryHekangkai 您是从哪里获得下载的apk的？
您能否分享与USB UVC设备一起使用的apk？

CarryHekangkai, 2020-09-11:
> @CarryHekangkai 您是从哪里获得下载的apk的？
> 您能否分享与USB UVC设备一起使用的apk？

是这个项目中提供的apk，我之前搞混了。
https://github.com/jiangdongguo/AndroidUSBCamera

paijwar, 2020-09-11:
@CarryHekangkai 感谢您分享这一点。 我也尝试过但无法从pgyer.com下载该应用程序，您可以通过将其上传到某个地方或直接在此处来帮助我使用apk。
我也尝试从源代码编译此文件，但从源代码无法正常工作，我的设备是Somagic EasyCapture 1c88：1001，它在device.c：944上失败
您是否遇到相同的问题？

CarryHekangkai, 2020-09-11:
> @CarryHekangkai 感谢您分享这一点。 我也尝试过但无法从pgyer.com下载该应用程序，您可以通过将其上传到某个地方或直接在此处来帮助我使用apk。
> 我也尝试从源代码编译此文件，但从源代码无法正常工作，我的设备是Somagic EasyCapture 1c88：1001，它在device.c：944上失败
> 您是否遇到相同的问题？

你可以给我你的邮箱

paijwar, 2020-09-11:
您可以通过paijwar与我联系[在gmail.com

CarryHekangkai, 2020-09-11:
> 您可以通过paijwar与我联系[在gmail.com
我不太懂你说的什么意思，你可以直接通过邮箱联系我：891210690@qq.com

jinglebiscuitsHL, 2021-07-21:
I encountered this error. There were 3 entries associated with one camera that was plugged into my device. I selected one of the other entries and this worked. No more error. Not sure why there are 3 entries for one camera though.

zaobin032411, 2022-05-20:
不解决了吗 作者 java.lang.UnsupportedOperationException: open failed:result=-50

network2006, 2022-05-31:
> I did downgrade the ndk version to 14b. still getting the error while opening camera.
> 
> Same code working on Moto G 5.0. but app crashing on samsung's all devices (I tried on Galaxy s4 and s5) 02-19 19:19:06.571 28018-28050/com.serenegiant.usbcameratest E/libuvc/device: [28050_device.c:944:uvc_scan_control]:UVC_ERROR_INVALID_DEVICE 02-19 19:19:06.571 28018-28050/com.serenegiant.usbcameratest E/UVCCamera: [28050_UVCCamera.cpp:172:connect]:could not open camera:err=-50 02-19 19:19:06.571 28018-28050/com.serenegiant.usbcameratest E/AndroidRuntime: FATAL EXCEPTION: BaseActivity Process: com.serenegiant.usbcameratest, PID: 28018 java.lang.UnsupportedOperationException: open failed:result=-50 at com.serenegiant.usb.UVCCamera.open(UVCCamera.java:204) at com.serenegiant.usbcameratest.MainActivity$2$1.run(MainActivity.java:142) at android.os.Handler.handleCallback(Handler.java:739) at android.os.Handler.dispatchMessage(Handler.java:95) at android.os.Looper.loop(Looper.java:145) at android.os.HandlerThread.run(HandlerThread.java:61)

Question ：This is my error：UVC_ERROR_INVALID_DEVICE not open camera:err=-50 ！！！
Answer ：mCameraHelper.requestPermission(0)是默认打开第一个usb设备，
                    但是实际上所使用的usb摄像头并未处于第一位，
                    因此将requestPermission(0)改为requestPermission(1)或其他数字后再进行测试
                    ，一般就可以解决问题了。

network2006, 2022-05-31:
> 不解决了吗 作者 java.lang.UnsupportedOperationException: open failed:result=-50

mCameraHelper.requestPermission(0)是默认打开第一个usb设备，
                    但是实际上所使用的usb摄像头并未处于第一位，
                    因此将requestPermission(0)改为requestPermission(1)或其他数字后再进行测试
                    ，一般就可以解决问题了。