saki4510t/UVCCamera #323: could not open camera:err=-1

sunnythree, 2018-02-23:
I encountered such a problem:When I use 'ndk-build' to compile native code with the command line，the code run well,however，when I clean project in android studio,and rebuild code in android studio,The native code compiled sucessfully,but when I run code on devices,There was such a mistake:'could not open camera:err=-1'? So ,Is there any difference between compile native code with command line and compile native code with android studio?

YongEun-vc, 2018-02-26:
Is the ndk used by Android Studio is different from that you used at command line?
Maybe #275 can be helpful.

shibbs, 2018-02-28:
This solved the issue for me, thank you @YongEun-vc