saki4510t/UVCCamera #329: green screen

ray0807, 2018-03-20:
[image]
[image]

1. My UVC camera can use in system camera well.
2.Change into my app (UVCCamera demo[usbcameratest5]),use UVC camera  , it shows green screen.
log:

[camera_0.log](https://github.com/saki4510t/UVCCamera/files/1827631/camera_0.log)

ray0807, 2018-03-26:
03-26 07:03:28.895 6316-6316/? I/art: Late-enabling -Xcheck:jni
03-26 07:03:28.932 6316-6316/com.serenegiant.usbcameratest0 D/DisplayManager: getDisplayInfo: displayId=0, info=DisplayInfo{"内置屏幕", uniqueId "local:0", app 1920 x 1032, real 1920 x 1080, largest app 1920 x 1847, smallest app 1080 x 1007, 66.583 fps, supportedRefreshRates [66.583], rotation 0, density 160 (159.89508 x 160.42105) dpi, layerStack 0, appVsyncOff 0, presDeadline 16018848, type BUILT_IN, state ON, FLAG_SECURE, FLAG_SUPPORTS_PROTECTED_BUFFERS}
03-26 07:03:28.935 6316-6316/com.serenegiant.usbcameratest0 D/DisplayManager: getDisplayInfo: displayId=0, info=DisplayInfo{"内置屏幕", uniqueId "local:0", app 1920 x 1032, real 1920 x 1080, largest app 1920 x 1847, smallest app 1080 x 1007, 66.583 fps, supportedRefreshRates [66.583], rotation 0, density 160 (159.89508 x 160.42105) dpi, layerStack 0, appVsyncOff 0, presDeadline 16018848, type BUILT_IN, state ON, FLAG_SECURE, FLAG_SUPPORTS_PROTECTED_BUFFERS}
03-26 07:03:28.940 6316-6316/com.serenegiant.usbcameratest0 D/DisplayManager: getDisplayInfo: displayId=0, info=DisplayInfo{"内置屏幕", uniqueId "local:0", app 1920 x 1032, real 1920 x 1080, largest app 1920 x 1847, smallest app 1080 x 1007, 66.583 fps, supportedRefreshRates [66.583], rotation 0, density 160 (159.89508 x 160.42105) dpi, layerStack 0, appVsyncOff 0, presDeadline 16018848, type BUILT_IN, state ON, FLAG_SECURE, FLAG_SUPPORTS_PROTECTED_BUFFERS}
03-26 07:03:28.942 6316-6316/com.serenegiant.usbcameratest0 D/DisplayManager: getDisplayInfo: displayId=0, info=DisplayInfo{"内置屏幕", uniqueId "local:0", app 1920 x 1032, real 1920 x 1080, largest app 1920 x 1847, smallest app 1080 x 1007, 66.583 fps, supportedRefreshRates [66.583], rotation 0, density 160 (159.89508 x 160.42105) dpi, layerStack 0, appVsyncOff 0, presDeadline 16018848, type BUILT_IN, state ON, FLAG_SECURE, FLAG_SUPPORTS_PROTECTED_BUFFERS}
03-26 07:03:29.114 6316-6316/com.serenegiant.usbcameratest0 I/InstantRun: starting instant run server: is main process
03-26 07:03:29.168 6316-6316/com.serenegiant.usbcameratest0 D/DisplayManager: getDisplayInfo: displayId=0, info=DisplayInfo{"内置屏幕", uniqueId "local:0", app 1920 x 1032, real 1920 x 1080, largest app 1920 x 1847, smallest app 1080 x 1007, 66.583 fps, supportedRefreshRates [66.583], rotation 0, density 160 (159.89508 x 160.42105) dpi, layerStack 0, appVsyncOff 0, presDeadline 16018848, type BUILT_IN, state ON, FLAG_SECURE, FLAG_SUPPORTS_PROTECTED_BUFFERS}
03-26 07:03:29.172 6316-6316/com.serenegiant.usbcameratest0 D/DisplayManager: getDisplayInfo: displayId=0, info=DisplayInfo{"内置屏幕", uniqueId "local:0", app 1920 x 1032, real 1920 x 1080, largest app 1920 x 1847, smallest app 1080 x 1007, 66.583 fps, supportedRefreshRates [66.583], rotation 0, density 160 (159.89508 x 160.42105) dpi, layerStack 0, appVsyncOff 0, presDeadline 16018848, type BUILT_IN, state ON, FLAG_SECURE, FLAG_SUPPORTS_PROTECTED_BUFFERS}
03-26 07:03:29.178 6316-6316/com.serenegiant.usbcameratest0 D/DisplayManager: getDisplayInfo: displayId=0, info=DisplayInfo{"内置屏幕", uniqueId "local:0", app 1920 x 1032, real 1920 x 1080, largest app 1920 x 1847, smallest app 1080 x 1007, 66.583 fps, supportedRefreshRates [66.583], rotation 0, density 160 (159.89508 x 160.42105) dpi, layerStack 0, appVsyncOff 0, presDeadline 16018848, type BUILT_IN, state ON, FLAG_SECURE, FLAG_SUPPORTS_PROTECTED_BUFFERS}
03-26 07:03:29.178 6316-6316/com.serenegiant.usbcameratest0 D/PhoneWindow: DecorView - SCREEN_WEITH = 1920 - SCREE_HEIGHT = 1032
03-26 07:03:29.178 6316-6316/com.serenegiant.usbcameratest0 I/System.out: action Title!
03-26 07:03:29.213 6316-6316/com.serenegiant.usbcameratest0 V/MainActivity: onStart:
03-26 07:03:29.218 6316-6316/com.serenegiant.usbcameratest0 D/DisplayManager: getDisplayInfo: displayId=0, info=DisplayInfo{"内置屏幕", uniqueId "local:0", app 1920 x 1032, real 1920 x 1080, largest app 1920 x 1847, smallest app 1080 x 1007, 66.583 fps, supportedRefreshRates [66.583], rotation 0, density 160 (159.89508 x 160.42105) dpi, layerStack 0, appVsyncOff 0, presDeadline 16018848, type BUILT_IN, state ON, FLAG_SECURE, FLAG_SUPPORTS_PROTECTED_BUFFERS}
03-26 07:03:29.219 6316-6316/com.serenegiant.usbcameratest0 D/DisplayManager: getDisplayInfo: displayId=0, info=DisplayInfo{"内置屏幕", uniqueId "local:0", app 1920 x 1032, real 1920 x 1080, largest app 1920 x 1847, smallest app 1080 x 1007, 66.583 fps, supportedRefreshRates [66.583], rotation 0, density 160 (159.89508 x 160.42105) dpi, layerStack 0, appVsyncOff 0, presDeadline 16018848, type BUILT_IN, state ON, FLAG_SECURE, FLAG_SUPPORTS_PROTECTED_BUFFERS}
03-26 07:03:29.224 6316-6337/com.serenegiant.usbcameratest0 D/OpenGLRenderer: Use EGL_SWAP_BEHAVIOR_PRESERVED: true
03-26 07:03:29.237 6316-6316/com.serenegiant.usbcameratest0 D/ViewRootImpl: onAttachToWindow register content observer attrs=WM.LayoutParams{(0,0)(fillxfill)mPosX=0mPosY=0mHScale=1.0mVScale=1.0 align=UNDEFINE taskId=28 sim=#100 ty=1 fl=#1810100 wanim=0x10303e5 surfaceInsets=Rect(0, 0 - 0, 0) needsMenuKey=2packageName=com.serenegiant.usbcameratest0token=android.os.BinderProxy@c5d7940}
03-26 07:03:29.254 6316-6316/com.serenegiant.usbcameratest0 D/ViewRootImpl: 1920<<<<<< BACK FROM relayoutWM.LayoutParams{(0,0)(fillxfill)mPosX=0mPosY=0mHScale=1.0mVScale=1.0 align=UNDEFINE taskId=28 sim=#120 ty=1 fl=#1810100 fmt=-3 wanim=0x10303e5 surfaceInsets=Rect(0, 0 - 0, 0) needsMenuKey=2packageName=com.serenegiant.usbcameratest0token=android.os.BinderProxy@c5d7940}
[... cut ...]
03-26 07:03:36.179 6316-6335/com.serenegiant.usbcameratest0 I/MainActivity: supportedSize:{"formats":[{"index":1,"type":4,"default":1,"size":["640x480","352x288","320x240","176x144","160x120"]}]}
03-26 07:03:36.184 6316-6353/com.serenegiant.usbcameratest0 I/libUVCCamera: [6353*UVCPreview.cpp:492:prepare_preview]:frameSize=(640,480)@YUYV
03-26 07:03:36.433 6316-6354/com.serenegiant.usbcameratest0 W/libUVCCamera: [6354*UVCPreview.cpp:119:get_frame]:allocate new frame
03-26 07:03:36.434 6316-6353/com.serenegiant.usbcameratest0 W/libUVCCamera: [6353*UVCPreview.cpp:119:get_frame]:allocate new frame

Erichthonius, 2018-11-09:
@ray0807 你好，你这个绿屏问题解决了吗

hjhcs121, 2019-09-04:
> @ray0807 你好，你这个绿屏问题解决了吗

你好，你知道绿屏问题怎么解决了吗？

thefaisalurrehman, 2022-10-21:
Do you find any solution?