saki4510t/UVCCamera #330: Zero Packet Transfer

heinrichu, 2018-03-26:
Hello,

someone knows about "zero packet" transfer issue? 
It means, the software it self is working, but no data is delivered by camera. So i have a blank screen, no preview.

I get an error message saying "zero packet (transfer)". 

I'm using an infrared UVC device, mostly working well with Android 6, on phone and tablet.
Switching to Android N on tablet and phone results in the following phenomena:
- the camera only works fine after an inital pressing of start and stop ... before, i receive zero packets errors.

The sample app of Saki just works fine with the cameras. So there should be a software solution for this.

Greetings

heinrichu, 2018-04-11:
OK. Solved by my self. Propietary camera requried setting of Altsetting 0 before setting 1. 
Android N has not done this on its own.

closed 2018-04-11 by heinrichu

WestbrookZero, 2019-05-14:
I also meet the same problem , No data is received , I don't understand how you slove it，Can you describe it more specifically by code?