saki4510t/UVCCamera #335: Linker issue with x86 arch with API>23

deval1491, 2018-04-10:
Issue seem to be with x86 arch only. Can anyone please solve this?

11-11 12:13:20.524  5231  5231 V UVCCameraTextureView: getPreviewTexture:
11-11 12:13:20.524  5231  5254 V UVCCameraTextureView: setVideoEncoder:
11-11 12:13:20.524  5231  5263 I UVCCameraTextureView: RenderThread#updatePreviewSurface:
11-11 12:13:20.525  5231  5263 V UVCCameraTextureView: updatePreviewSurface:tex_id=1
11-11 12:13:20.528  4301  4371 W _SHM    : IasAvbAudioShmProvider::copyJob(853):[avb_eavb_hmi_p] tx buffer overrun written = 0 expected = 192
11-11 12:13:20.529  5231  5254 E linker  : "/data/app/com.serenegiant.opencvwithuvc-t32Y0_f3i8eA6GzsFevgGQ==/lib/x86/libjpeg-turbo1500.so" has text relocations **(https://android.googlesource.com/platform/bionic/+/master/android-changes-for-ndk-developers.md#Text-Relocations-Enforced-for-API-level-23)**
11-11 12:13:20.532  4301  4371 W _SHM    : IasAvbAudioShmProvider::copyJob(853):[avb_eavb_hmi_p] tx buffer overrun written = 0 expected = 192

acaliaro, 2018-10-29:
I have the same problem. There is a solution for this?

acaliaro, 2018-10-30:
@saki4510t Could you give some suggestions about this problem?

acaliaro, 2018-10-30:
you should add this rows

`LOCAL_ASMFLAGS += -DPIC
`
before the row

`else ifeq ($(TARGET_ARCH_ABI),mips)
`
in libjpeg-turboXXX\android.mk