saki4510t/UVCCamera #337: How I draw a line on every frames ?

805447391, 2018-04-11:
Same as titile.
I want to get per frame when previewing, and do something with this frame(like face detection or recognize), finally render to UVCCameraTextureView.
Can you give me a solution? I am very pleasure if you can reply my issue, thanks sincerely.

805447391, 2018-04-18:
reply myself:
```
uvcCamera.setFrameCallback(new IFrameCallback() {
            private Bitmap finalBitmap;
            private byte[] bufferAvoidGC;

            @Override
            public void onFrame(ByteBuffer frame) {
                     // some operation
            }
        }, UVCCamera.PIXEL_FORMAT_RGBX);
```

ritchie-w, 2018-05-15:
Hi,

I would like to print a text in the shown video, but the function is never called.
I also would like to have the additional text shown in the recorded video.
Is the also done within this function ?

I am using the following code:

```
		private final IFrameCallback mIFrameCallback = new IFrameCallback() {
		@Override
		public void onFrame(final ByteBuffer frame) {
			frame.clear();
			synchronized (bitmap) {
				bitmap.copyPixelsFromBuffer(frame);
				Canvas c = new Canvas(bitmap);
				Paint p = new Paint();
				p.setColor(Color.rgb(61, 61, 61));
				p.setTextSize((int) (14 * 10));
				p.setShadowLayer(1f, 0f, 1f, Color.WHITE);
				c.drawText("Hello", 30, 210, p);
			}
		}
	};

```
The function is never call, because I set a break point which never reached.

Any hints ?

Best regards
R.

rgm-22, 2021-10-02:
Hi @ritchie-w!

Did you solve it?

Thank you very much in advance!