saki4510t/UVCCamera #339: Camera Service does not run after refactoring the project usbCameraTest4

ritchie-w, 2018-04-14:
Hi,

I refactoring the testprogram "usbCameraTest4" to my own project name in the following way,
"Project"-Master

-"Project" 
-------../Java
-------------com
-----------------"Project"
-----------------serenegiant
--------------------service
--------------------serviceclient
-------../res
- libuvccamera
- usbCameraCommon

The building works fine, but the application does not start the "CameraServer", so 
no stream is shown.

Where is my failure ?
Best regards
R.

ritchie-w, 2018-04-15:
I found the reason in the AndroidManifest.xml.

                <action android:name="com.<Project>.service.IUVCService" />
                <action android:name="com.<Project>.service.IUVCSlaveService" />

Sorry. Is my first Android project.

closed 2018-04-15 by ritchie-w

ADUOH2017, 2018-04-19:
Hi ritchie-w,

I am also a begginner in android deveelopper and I have a problem with usbCameraTest4.

First of all,  it compiled correctly and installed in my ohone but it is crashed and  I don't know what is the problem.

Can you help me please.
It is very urgent.
Thank you very much in advance