saki4510t/UVCCamera #343: memory leak

qiqisteve, 2018-04-19:
[image]

i  have modified the usbcameraTest0 project ,make it polling multiple camera, each camera shows server seconds, finally i found that android profiler memory is always growing ,even if i close camera each turns

i have found there is exits another post about memory leak,but not woks,

this is my code snippet:

`	private final OnClickListener mOnClickListener = new OnClickListener() {
		@Override
		public void onClick(final View view) { 
			if(mDevicesList.size() > 0 )
			{
				mHandler.postDelayed(new Runnable() {
					@Override
					public void run() {
						int t = index%mDevicesList.size();
						openUsbCamera(mDevicesList.get(t).getDeviceName());
						mHandler.postDelayed(this,5000);
						index++;
					}
				},1000);
			}
		}
	};

	private void openUsbCamera(final String deviceName)
	{
		synchronized (mSync) {
			if (mUVCCamera != null) {
				mUVCCamera.stopPreview();
			 	mUVCCamera.close();
				mUVCCamera.destroy();

				isActive = isPreview = false;
			}
		}
		queueEvent(new Runnable() {
			@Override
			public void run() {
				synchronized (mSync) {
					 UVCCamera camera;

					camera = new UVCCamera();
					camera.open(mDevicesControlBlockList.get(deviceName));

					if (DEBUG) Log.i(TAG, "supportedSize:" + camera.getSupportedSize());
					try {
						camera.setPreviewSize(UVCCamera.DEFAULT_PREVIEW_WIDTH, UVCCamera.DEFAULT_PREVIEW_HEIGHT, UVCCamera.FRAME_FORMAT_MJPEG);
					} catch (final IllegalArgumentException e) {
						try {
							// fallback to YUV mode
							camera.setPreviewSize(UVCCamera.DEFAULT_PREVIEW_WIDTH, UVCCamera.DEFAULT_PREVIEW_HEIGHT, UVCCamera.DEFAULT_PREVIEW_MODE);
						} catch (final IllegalArgumentException e1) {
							camera.destroy();
							return;
						}
					}
					mPreviewSurface = mUVCCameraView.getHolder().getSurface();
					if (mPreviewSurface != null) {
						isActive = true;
						camera.setPreviewDisplay(mPreviewSurface);
						camera.startPreview();
						isPreview = true;
					}
					mCameras.put(deviceName,camera);
					mUVCCamera = camera;

				}
			}
		}, 0);
	}
`

qiqisteve, 2018-04-19:
@saki4510t pls help me

paipeng, 2018-05-03:
Hello,

I have also problem by stop/restart camera for showing a result image.

At the 7th time stop/restart camera, the app crashed.

BarryAlertPan, 2018-05-07:
I also encountered this problem, how do you solve it?

boboqun, 2020-08-12:
same problem， please help me.