saki4510t/UVCCamera #345: YUV420 not supported

Harishvats, 2018-04-27:
I have a custom UVC camera which gives video in YUV420 format.
I get no preview in this case. 
It gives error "could not negotiate with camera errpor=-51" . 
What can be done to support this format also?

heinrichu, 2018-05-16:
Hello,

the error could also be caused by other reasons.
At first you need to set set the correct frame dimensions, (maximum) frame rate, preview mode (here probably to `0` for YUV), etc. The default dimension is VGA(640x480). If you cam does not provide this, a preview will not be possible. (have alook into `UVCCamera.java: setPreviewSize` and the defined constants at the beginning of the code file)

I also use a custom camera and i was finally able to fix my problems by selecting the appropriate parameters.

If your camera also uses a special color format, you could try to edit the native code (which i did).
Try to find the function in native code file `void UVCPreview::do_preview(uvc_stream_ctrl_t *ctrl)`
and adapt conversion function for your camera's format.
The c-file frame.c offers many different functions handling many color formats.

Greetings

heaveninfosoft, 2020-06-27:
> Hello,
> 
> the error could also be caused by other reasons.
> At first you need to set set the correct frame dimensions, (maximum) frame rate, preview mode (here probably to `0` for YUV), etc. The default dimension is VGA(640x480). If you cam does not provide this, a preview will not be possible. (have alook into `UVCCamera.java: setPreviewSize` and the defined constants at the beginning of the code file)
> 
> I also use a custom camera and i was finally able to fix my problems by selecting the appropriate parameters.
> 
> If your camera also uses a special color format, you could try to edit the native code (which i did).
> Try to find the function in native code file `void UVCPreview::do_preview(uvc_stream_ctrl_t *ctrl)`
> and adapt conversion function for your camera's format.
> The c-file frame.c (libuvc.h) offers many different functions handling many color formats.
> 
> Greetings

hey i got  
E/libUVCCamera: [3386*UVCPreview.cpp:507:prepare_preview]:could not negotiate with camera:err=-51

can you help me??

B3rn4rdDL, 2022-05-04:
Hey,
I got the same error, did you succeed with your custom camera ? Mine gives video in YUV422 format and I can't get a preview due to this error : E/libUVCCamera: [3386*UVCPreview.cpp:517:prepare_preview]:could not negotiate with camera:err=-51

anmorick, 2024-08-19:
Did you check the range of fps? if fps lower or higher  then default , it will be show camera:err=-51
So make sure your camera config.