saki4510t/UVCCamera #346: Orientation

xwilus, 2018-04-29:
Hi,

First of all I would like to say thank you very much for the library. It works awesomely. Took me quick to implement it and use it.

I would just like to know something. I have read up quite a lot about the orientation for the streaming. When I start up on the tablet I test it works fine but on the another tablet the activity layout is presented the right way up but the actual stream on the surfaceview is upside down. What would be the best way to make it stream right way up ? 

Regards,
Willie

UbuntuLover, 2018-05-02:
Do you know where to get the yuv data without using the cameraviewinterface?

xwilus, 2018-05-03:
No I do not. I have only recently started using these libraries