saki4510t/UVCCamera #350: 开启关闭问题

zzshuihu, 2018-05-07:
当相机开启之后，退出，然后再开启预览，就会出现无法预览的情况,程序一直再waitPreviewFrame()等待，退出后，再次开启，相机就会开启不了，需要重新拔插USB，请问该怎么解决啊？

anoop-sharma, 2018-05-08:
This only happens if you are not closing camera properly. In my application I am doing like this

onCreate : Camera is created

onStart : Camera is started

onStop : Camera Stopped

onPause : No Changes

onResume : Check and resume camera

onDestroy : Camera Stopped and destroyed

You have to be very careful if your app crashes while camera is open then it is not able to reattach. You need to re-plug the camera. Following is the code if it helps

protected void onCreate(final Bundle savedInstanceState) {
        /** Camera Code **/
        mUVCCameraView = (SurfaceView) findViewById(R.id.camera_surface_view);
        mUVCCameraView.getHolder().addCallback(mSurfaceViewCallback);

        mUSBMonitor = new USBMonitor(this, mOnDeviceConnectListener);
    }

    @Override
    protected void onStart() {
        super.onStart();
        synchronized (mSync) {
            if (mUSBMonitor != null) {
                mUSBMonitor.register();
            }
        }
    }

    @Override
    protected void onStop() {
        synchronized (mSync) {
            if (mUSBMonitor != null) {
                mUSBMonitor.unregister();
            }
        }
        super.onStop();
    }

@Override
    protected void onDestroy() {
        closeApp();
        super.onDestroy();
    }

@Override
    protected void onPause() {
        stopStreamingServer();
        super.onPause();
    }

    @Override
    protected void onResume() {
        startApp();
        super.onResume();
    }

protected void closeApp() {
        synchronized (mSync) {
            isActive = isPreview = false;
            if (mUVCCamera != null) {
                mUVCCamera.destroy();
                mUVCCamera = null;
            }
            if (mUSBMonitor != null) {
                mUSBMonitor.destroy();
                mUSBMonitor = null;
            }
        }
        mUVCCameraView = null;
    }

    protected void startApp() {
        if (mUSBMonitor != null) {
            mUSBMonitor.register();
        }
    }

Please check if this helps you.

zzshuihu, 2018-05-09:
thanks  brother!

bingyanxueling, 2018-06-13:
我也出现了同样的问题，请问您解决了吗？

bingyanxueling, 2018-06-13:
上面的解决方法对我没有作用

16rd, 2018-08-31:
有解决这个bug的同学么？

zhoucater, 2020-06-11:
这个问题其实非常奇怪，设计到前程同步，有个地方的判断去掉就能解决问题，即使摄像头打开失败也有预览出现public void handleStartPreview(final Object surface) {
			if (DEBUG) Log.v(TAG_THREAD, "handleStartPreview:");
			if ((mUVCCamera == null) || mIsPreviewing) return;
。。。。。。
}
mIsPreviewing 把这个判断去掉即可。