saki4510t/UVCCamera #354: Crashing:  java.lang.NoSuchMethodError: no static or non-static method "Lcom/serenegiant/usb/UVCCamera;.nativeSetStatusCallback(JLcom/serenegiant/usb/IStatusCallback

MADHUSUDHANRAO527, 2018-05-09:
I am trying to integrated UVC camera in my app. I am getting this crash log.Can any one help me out? 

[image]

FATAL EXCEPTION: USBMonitor
                                                                     Process: com.android.launcher3, PID: 1741
                                                                     java.lang.NoSuchMethodError: no static or non-static method "Lcom/serenegiant/usb/UVCCamera;.nativeSetStatusCallback(JLcom/serenegiant/usb/IStatusCallback;)I"
                                                                         at java.lang.Runtime.nativeLoad(Native Method)
                                                                         at java.lang.Runtime.doLoad(Runtime.java:1099)
                                                                         at java.lang.Runtime.loadLibrary0(Runtime.java:1014)
                                                                         at java.lang.System.loadLibrary(System.java:1657)
                                                                         at com.serenegiant.usb.UVCCamera.<clinit>(UVCCamera.java:122)
                                                                         at com.bosch.mykie.MykieHomeActivity$2.onConnect(MykieHomeActivity.java:635)
                                                                         at com.serenegiant.usb.USBMonitor$3.run(USBMonitor.java:571)
                                                                         at android.os.Handler.handleCallback(Handler.java:789)
                                                                         at android.os.Handler.dispatchMessage(Handler.java:98)
                                                                         at android.os.Looper.loop(Looper.java:164)
                                                                         at android.os.HandlerThread.run(HandlerThread.java:65)

MADHUSUDHANRAO527, 2018-05-10:
After a lot of struggles i have solved my problem by my own 

The issue for me was with pro guard.
I resolved it by adding
-keep class com.serenegiant.** { *; }

to the pro guard rules file.

hope that helps someone

closed 2018-05-10 by MADHUSUDHANRAO527

cenmmy, 2020-09-09:
thank u, i solved my problem by your method