saki4510t/UVCCamera #355: App fails when trying to display images from Logitech's C270 camera

hcpiteira, 2018-05-11:
Hi there!

I've been trying to preview images from a Logitech C270 camera, by connecting it to the Samsung A3 2016 device, via micro-USB port. In order to try the app I built the camera test example apk's in Android Studio, and installed them on the device, the app runs fine, detects the camera when plugged, but when I select "OK" to open it, is simply says "USBCameraTesteX" has stopped, and closes automatically. 

I've already tried the following options, as suggested in #275 and #181 issues, but none of these have worked for me:
- Downgrade my NKD (I was using NDK17 and changed to NDK14b);
- Tried changing the compiler to gcc by uncomment "NDK_TOOLCHAIN_VERSION := 4.9" from "libuvccamera/src/main/jni/Application.mk"

Has anyone had this problem before?
I would really appreciate your help, as I am new to Android and am still trying to get the hang of it.

Thanks in advance!

heinrichu, 2018-05-16:
Hello,

you could start a debug session with Android Studio in order to find the error. Usually this workes pretty well.

Use a wifi-debug session to debug and plugin your cam via your single USB-port simultaneously. 
(maybe a secondary USB-Hub is working as well?)
Just use Google for finding a manual about wifi-debugging.

Greetings

hcpiteira, 2018-05-17:
Hi @heinrichu , first of thank you for your response! I can't use my smartphone in developers mode for debugging right now though...

But I noticed something else, there seems to be some problem with JNI function, since I've got several error messages in the UVCCamera file  `/UVCCamera/libuvccamera/src/main/java/com/serenegiant/usb/UVCCamera.java`  from line 1069 to 1225, saying that JNI functions are missing. Have you got any idea on why this is happening or how to solve it?

Maybe this is the problem that is causing the app to fail when opening the camera.

heinrichu, 2018-05-18:
Hello,

you might want to show this error messages to us? :grimacing:
Don't be shy, just paste it in a comment. :smile: 

The project should work fine if you compile and run it.
I tested usbCameraTest8 and run it on Samsung A3 2017 .... and there seems not be a problem at all.

You used the debug version for creating apk? Haven't tested it with release.
Did you already made some modification of code?

Is there a technical problem you can't debug with your phone?

Greetings
H.

hcpiteira, 2018-05-18:
Hello again @heinrichu , thank you so much for your help, I didn't change the code, but I tried rebuilding the project once again, and it just worked! :)

It seems that the problem wasn't with the app, but with another app I had build and installed on my smartphone before I tried UVCCamera, since I gave permissions to that app to always open the logitech C270 camera if it was connected via micro USB, therefore I believe the app crashed because the camera was already in use by my other app. Either it was that, or I just did something wrong building the UVCCamera app previously.

Either way, thanks again, you were very helpful!

Greetings,
Helena

closed 2018-05-18 by hcpiteira