saki4510t/UVCCamera #357: How to improve the frame rate when preview in 1080P?

UbuntuLover, 2018-05-11:
AS the title, I want to try preview in 1080P with 30FPS. Could any offer some instructions?

1159658610, 2019-04-29:
Have you solved the problem?  I hope to get your help.   The sincerity anticipates your reply .

UbuntuLover, 2019-04-30:
> Have you solved the problem? I hope to get your help. The sincerity anticipates your reply .

Sorry to told you I have not solve this issue... :( But for my personal advice you can refer others suggestions like rewrite the whole process and try to trace the data flow... If you have any ideas, please let me know.

1159658610, 2019-06-21:
> Sorry to told you I have not solve this issue... :( But for my personal advice you can refer others suggestions like rewrite the whole process and try to trace the data flow... If you have any [...]

thank you for response

UbuntuLover, 2019-06-23:
@1159658610  No worries.