saki4510t/UVCCamera #358: Process 'command 'D:\SDK\ndk-bundle/ndk-build.cmd'' finished with non-zero exit value 2

12ndxu, 2018-05-14:
after I click run ,logcat output this message,how can I fix it?

iskander-saitbatalov, 2018-05-14:
Try to remove previous build and rebuild project.

12ndxu, 2018-05-15:
if somebody fix this problem，tell me please

heinrichu, 2018-05-15:
Hi,

I have the same problem. It turns out it is a problem with new NDK (17.0.4754217).
With 16.1.4479499 is is working. Also with the "famous" NDK14b which is in focus of many posts here.
So change to other NDK. You can download the older versions and link it in "local.properties" file of root directory.

for example: 

`## This file is automatically generated by Android Studio.`
`# Do not modify this file -- YOUR CHANGES WILL BE ERASED!`
`#`
`# This file must *NOT* be checked into Version Control Systems,`
`# as it contains information specific to your local configuration.`
`#`
`# Location of the SDK. This is only used by Gradle.`
`# For customization when using a Version Control System, please read the`
`# header note.`
`#Mon May 07 16:47:37 CEST 2018`
`ndk.dir=C\:\\Users\\<USER>\\android-ndk-r14b`
`sdk.dir=C\:\\Users\\<USER>\\AppData\\Local\\Android\\Sdk`

heinrichu, 2018-05-18:
OK....i just found out the problem.
NDK 17 does not support target architectures armeabi and mips anymore...as pointed out in #277  
So if you want to use NDK17, remove entries in `libuvccamera\src\main\jni\Application.mk`
Change:
```
APP_ABI := armeabi armeabi-v7a x86 mips
```
to
```
APP_ABI := armeabi-v7a x86
```
Close Android Studio.
Then you need to delete the content of the lib directories in `libuvccamera\src\main\libs\*\*.so`.
If not you get probably an additional compiler error.
Open Android Studio an rebuild your project-/s.
At least it works for me.

m9r1a, 2018-05-20:
I also have this problem.
my environment:

gradle ver.:3.3
Android plugin ver.:2.3.1
Android Studio 3.1.2
Build #AI-173.4720617, built on April 14, 2018
JRE: 1.8.0_152-release-1024-b02 amd64
JVM: OpenJDK 64-Bit Server VM by JetBrains s.r.o
Windows 10 10.0

I check different version of NDK; 13,14,16,17

local.properties:

> ndk.dir=F\:\\android-ndk-r14b
> sdk.dir=F\:\\Sdk

any help please...

heinrichu, 2018-05-22:
What about changing:
`libuvccamera\src\main\jni\Application.mk`

Why dont you use the standard paths in local.properties?
You installed SDK and NDK via Android Studio?

Whats the full error message?

m9r1a, 2018-05-22:
I used windows; but now I'm using Ubuntu and everything is ok.

xslandlr, 2018-06-27:
I connect camera with hub。a  hub connect 10 camera，when  i  reboot  tablet ，UsbDevice.getDeviceId() ,The deviceId  is  different everyTime。I  hope when  i reboot the tablet，i get  the  unique  Id。what  shuld  i  do?

Eido95, 2019-03-26:
@heinrichu comment worked for me.

However, I used the following line:
```
APP_ABI := all
```

leexinyi, 2019-04-01:
I also have the same problem,but my NDK verdion is android-ndk-r10e. and the bug is <**org.gradle.api.tasks.TaskExecutionException: Execution failed for task ':app:ndkBuild'**>
then I use gradle ndkBuild,
<**FAILURE: Build failed with an exception.

* Where:
Build file '/home/tao/GrRxFm/build.gradle' line: 7

* What went wrong:
A problem occurred evaluating root project 'GrRxFm'.
> Could not find method google() for arguments [] on repository container.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

BUILD FAILED**>

however, when I delete

<**buildscript {
    repositories {
        jcenter()
        maven {url 'https://maven.google.com'}
       // google()
in build.gradle,and make build again,
the error is 
buildscript {
    repositories {
        jcenter()
        maven {url 'https://maven.google.com'}
       // google()**>
the error is 
<**org.gradle.process.internal.ExecException: Process 'command '/home/tao/Android/Sdk/ndk-bundle/ndk-build'' finished with non-zero exit value 2**>

I cd into ~/Android/Sdk/ndk-bundle
and ./ndk-build
it says
<Android NDK: APP_PLATFORM not set. Defaulting to minimum supported version android-16. 
make: *** No rule to make target `GrAndroid.mk'.  Stop.>

I try different version of gradle. and use the true path in /etc/profile . 

what I want to do is to run the project from https://codeload.github.com/trondeau/GrRxFM/zip/master

kwaithaka, 2019-05-24:
> OK....i just found out the problem.
> NDK 17 does not support target architectures armeabi and mips anymore...as pointed out in #277
> So if you want to use NDK17, remove entries in [...]

Worked for me.

whitecloudy, 2019-09-30:
I fixed this problem by this document. Apparently NDK which is installed by Android Studio is not working for previous NDK codes. So I just installed an older version and manually linked it with "Open Module Settings -> SDK Location"

https://stackoverflow.com/questions/29605029/android-studio-failed-build-ndk-project-non-zero-exit-value

TronQuick, 2019-12-05:
same as u ,  i have try all the methods from previos guys said , but them didn't work for me.help~~TAT

i m using Android Studio 3.5.2 , SDK API 28 , NDK 14(i have try ndk14/ndk16)
this is my build error report:
`org.gradle.execution.MultipleBuildFailures: Build completed with 1 failures.
at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
at java.lang.Thread.run

Caused by: org.gradle.api.tasks.TaskExecutionException: Execution failed for task ':libuvccamera:ndkBuild'.
... 6 more

Caused by: org.gradle.process.internal.ExecException: Process 'command 'C:\Users\Administrator\AppData\Local\Android\Sdk\ndk14/ndk-build.cmd'' finished with non-zero exit value 2
at java.util.Optional.orElseGet(Optional.java:267)
at java.util.Optional.map(Optional.java:215)
... 38 more`

leexinyi, 2019-12-09:
I  have tried ndk11e and gradle 4.4 with 3.1.3, it works for memaybe you don't set the true ndk. and this is my  configuration

and add "exclude 'META-INF/DEPENDENCIES'" to you build.gradle
 

------------------

李昕怡

北京邮电大学/本科生/信息与通信工程学院

18811371655

北京

 
 
 
------------------ Original ------------------
From: "TronQuick"; 
Date: 2019年12月5日(星期四) 下午2:16
To: "saki4510t/UVCCamera"; 
Cc: "leexinyi"; "Comment"; 
Subject: Re: [saki4510t/UVCCamera] Process 'command 'D:\SDK\ndk-bundle/ndk-build.cmd'' finished with non-zero exit value 2 (#358)

 

same as u ,  i have try all the methods from previos guys said , but them didn't work for me.help~~TAT
 
i m using Android Studio 3.5.2 , SDK API 28 , NDK 14(i have try ndk14/ndk16)
 this is my build error report:
 `org.gradle.execution.MultipleBuildFailures: Build completed with 1 failures.
 at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
 at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
 at java.lang.Thread.run
 
Caused by: org.gradle.api.tasks.TaskExecutionException: Execution failed for task ':libuvccamera:ndkBuild'.
 ... 6 more
 
Caused by: org.gradle.process.internal.ExecException: Process 'command 'C:\Users\Administrator\AppData\Local\Android\Sdk\ndk14/ndk-build.cmd'' finished with non-zero exit value 2
 at java.util.Optional.orElseGet(Optional.java:267)
 at java.util.Optional.map(Optional.java:215)
 ... 38 more`

TronQuick, 2019-12-09:
make sure that u'r using the right version of NDK , i fix it by using the ndk14_64 , gradle 3.5.0

zipswich, 2022-01-15:
> @heinrichu comment worked for me.
> 
> However, I used the following line instead, to target [all ABIs](https://developer.android.com/ndk/guides/abis#gc):
> 
> ```
> APP_ABI := all
> ```

Thank you all.  This is the final configuration that has worked for me:
Application.mk:
`APP_ABI := all`

local.properties:
```
ndk.dir=...\\AppData\\Local\\Android\\Sdk\\ndk\\17.2.4988734
sdk.dir=...\\AppData\\Local\\Android\\Sdk
```
build.gradle:
```
        compileSdkVersion 31
        buildToolsVersion '30.0.2'
        minSdkVersion 23
	argetSdkVersion 27
```
Android Studio Arctic Fox | 2020.3.1 Patch 4
Build #AI-203.7717.56.2031.7935034, built on November 21, 2021

the output aar's jni folder has the following:
[image]

Tofu404, 2023-02-24:
i new a empty project and than copy the libuvccamera model file to the new project，after that config the build.gradle, it work! 
[image]
[image]
[image]