saki4510t/UVCCamera #360: How can I add Text in Stream for recording and display ...

ritchie-w, 2018-05-21:
Hi,

I am working with the camera4 Sample and try to get a "date and time stamp" into the screen and the recorded mp3 file and did not found a way to do it.

https://github.com/saki4510t/UVCCamera/wiki/howto_create_bitmap_using_iframecallback
Shows only a way to display a text on the screen. I tried this with UVCCameraTest and it works fine.
But this is still not what I want.

Can somebody give me a hint for starting (or a small sample code).
I tried to modified UVCPreview.cpp, but I failed.

Best regards
R.

heinrichu, 2018-05-25:
Hi,

The frames arrive as bytebuffer from native lib in` AbstractCameraHandler:607` in callback function 
`public void onFrame(final ByteBuffer frame)`.

For record you might do the following:
If frame would be a bitmap, you could simply draw text inside.
So you might put the data into a bitmap and draw text to it...then get the data out of the bitmap and forward it to the encoding object. But im not aware about the performance of importing and exporting to a bitmap or if it works properly at all.

You could try something like this. Unfortunately im not aware of the color format which is returned.
Here I used for example RGB565. ... so check  for color format.

For Bitmap you can have the following formats: 
`Bitmap.Config.ALPHA_8,
Bitmap.Config.ARGB_4444,
Bitmap.Config.ARGB_8888,
Bitmap.Config.RGB_565`

``` javascript 
Bitmap bmp = Bitmap.createBitmap(width, height, Bitmap.Config.RGB_565);
bmp.copyPixelsFromBuffer(frame.asIntBuffer());
Paint paint = new Paint(Paint.DEV_KERN_TEXT_FLAG);
paint.setColor(Color.rgb(110,110, 110));
paint.setTextSize((int) (24));
Canvas canvas = new Canvas(bmp);
canvas.drawText(String.format("your date string"), paint);
```

Finally you have to get the bytebuffer from the bmp object.

For live view I would simply add a text view on top of the surface. Set text of label with your desired date resp. update the text frequently with the actual date or with the date of arriving frames.

Greetings

H.

ritchie-w, 2018-05-25:
Hi,

I already tried to modify the original frame, but I failed also. I add a ImageView 
for displaying the modified bitmap.

```
	// if you need frame data as byte array on Java side, you can use this callback method with UVCCamera#setFrameCallback
	// if you need to create Bitmap in IFrameCallback, please refer following snippet.
	final Bitmap bitmap = Bitmap.createBitmap(UVCCamera.DEFAULT_PREVIEW_WIDTH, UVCCamera.DEFAULT_PREVIEW_HEIGHT, Bitmap.Config.RGB_565);
	private final IFrameCallback mIFrameCallback = new IFrameCallback() {
		@Override
		public void onFrame(ByteBuffer frame) {
			frame.clear();
			synchronized (bitmap) {
				bitmap.copyPixelsFromBuffer(frame);
				Canvas c = new Canvas(bitmap);
				Paint p = new Paint();
				// text color - #3D3D3D
				p.setColor(Color.rgb(61, 61, 61));
				// text size in pixels
				p.setTextSize((int) (28));
				// text shadow
				p.setShadowLayer(1f, 0f, 1f, Color.WHITE);
				c.drawText("Hello ", 10, 220, p);
				ByteArrayOutputStream stream = new ByteArrayOutputStream();
				bitmap.compress(Bitmap.CompressFormat.PNG, 100, stream);

				frame.clear();
			}
			mImageView.post(mUpdateImageTask);
		}
	};
	
	private final Runnable mUpdateImageTask = new Runnable() {
		@Override
		public void run() {
			synchronized (bitmap) {
				mImageView.setImageBitmap(bitmap);
			}
		}
	};

```
The Image view shows the additional text on the screen. But the original frame was still unchanged.
I tried this in the "usbcameraTest" Modul.

Because of the command "frame.Clear" the original screen should show nothing. I get the impression, that the function OnFrame() will get a copy of the original frame, change to the format, what you selected. I also removed the "final" key word, nothing happens.

I also guess, that I have to do the same change in cameratest4 in the module "cameraserver,java" as a  onFrame() callbackfunction, when it works correct in the usbcameraTest.
 

Best regards
R.

ritchie-w, 2018-06-02:
Hello HeinrichU,

sorry, that was a misunderstanding from my side.
I tried the onframe () in the module (cameraserver).

Now I will try your solution.

Best regards
R.

ritchie-w, 2018-06-14:
Hello,

does somebody has a solution for this problem ? 

Best regards
R.

rgm-22, 2021-10-02:
Hi @ritchie-w!

Did you solve it?

Thank you very much in advance!

ritchie-w, 2021-10-03:
Hello rgm-22,
I am sorry. I did not get the function to run. Also it was to slow for my application as well.
Therefore i gave up.