saki4510t/UVCCamera #366: [android studio] compilation failed when modify compile by implementation

stephGitUName, 2018-05-31:
I'm working with Android studio 3.1.2 and the Java compiler says that "configuration 'compile' is obsolete and has been replaced whith 'implementation' and 'api'. The 'compile' option will be disable at the end of the year. 
My project which using UVCCamera source compile perfectly with the 'compile' option. However when I change 'compile' for 'implementation', my android project doe not compile any more with the following errors
"error: package com.serenegiant.glutils does not exist"
"error: cannot find symbol class GLDrawer2D"
Does anyone experiments this error too and has a solution ? 

In a nutshell:

Compile is successfull with this usbCameraCommun gradle file dependencies : 
dependencies {
    implementation fileTree(dir: 'libs', include: ['*.jar'])
    compile("com.serenegiant:common:${commonLibVersion}") {
        exclude module: 'support-v4'
    }
    implementation project(':libuvccamera')
}

and failed with these dependencies

dependencies {
    implementation fileTree(dir: 'libs', include: ['*.jar'])
    implementation("com.serenegiant:common:${commonLibVersion}") {
        exclude module: 'support-v4'
    }
    implementation project(':libuvccamera')
}

Thanks in advance

ritchie-w, 2018-05-31:
Try the key word "api". It works for me.

stephGitUName, 2018-06-01:
It works! Great! I will had a look at the differences between the two compile options 'compile' and 'api' in android studio.
Thanks!

closed 2018-06-01 by stephGitUName

RedwanSharafatKabir, 2023-12-03:
> It works! Great! I will had a look at the differences between the two compile options 'compile' and 'api' in android studio. Thanks!

I am getting this error - Caused by: org.gradle.internal.resolve.ModuleVersionNotFoundException: Could not find com.serenegiant:common:2.12.5.

How to solve this issue?