saki4510t/UVCCamera #367: Zoom functionality not working

shashankjamdade, 2018-06-04:
I am developing the android app having external USB-camera functinality. I want to integrate zoom functionality but 'setZoom(final int zoom)' function which is available in library is not working properly. I am setting different int values & I tried to print the current zoom value it always return me -11 . I'm not able to see any changes on zoom in my camera view. please help me as soon as possible .

shashankjamdade, 2018-06-14:
I managed to solve the problem (just replace the following code with your existing class, you will get resultant zoomed image in getImageBitmap() method just call it from your activity)

**UVCCameraTextureView.java**
/*
 *  UVCCamera
 *  library and sample to access to UVC web camera on non-rooted Android device
 *
 * Copyright (c) 2014-2017 saki t_saki@serenegiant.com
 *
 *  Licensed under the Apache License, Version 2.0 (the "License");
 *  you may not use this file except in compliance with the License.
 *   You may obtain a copy of the License at
 *
 *      http://www.apache.org/licenses/LICENSE-2.0
 *
 *   Unless required by applicable law or agreed to in writing, software
 *   distributed under the License is distributed on an "AS IS" BASIS,
 *   WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 *   See the License for the specific language governing permissions and
 *   limitations under the License.
 *
 *  All files in the folder are under this Apache License, Version 2.0.
 *  Files in the libjpeg-turbo, libusb, libuvc, rapidjson folder
 *  may have a different license, see the respective files.
 */

package com.serenegiant.usb.widget;

import android.content.Context;
import android.content.res.TypedArray;
import android.graphics.Bitmap;
import android.graphics.Canvas;
import android.graphics.Matrix;
import android.graphics.Paint;
import android.graphics.PointF;
import android.graphics.SurfaceTexture;
import android.hardware.Camera;
import android.hardware.camera2.CameraDevice;
import android.hardware.camera2.CaptureRequest;
import android.media.MediaRecorder;
import android.net.Uri;
import android.os.Build;
import android.os.Environment;
import android.os.Handler;
import android.os.Looper;
import android.os.Message;
import android.support.annotation.RequiresApi;
import android.util.AttributeSet;
import android.util.DisplayMetrics;
import android.util.Log;
import android.view.MotionEvent;
import android.view.ScaleGestureDetector;
import android.view.Surface;
import android.view.TextureView;
import android.view.View;

import com.jiangdg.libusbcamera.R;
import com.serenegiant.glutils.EGLBase;
import com.serenegiant.glutils.GLDrawer2D;
import com.serenegiant.glutils.es1.GLHelper;
import com.serenegiant.usb.IZoomCallback;
import com.serenegiant.usb.UVCCamera;
import com.serenegiant.usb.encoder.IVideoEncoder;
import com.serenegiant.usb.encoder.MediaEncoder;
import com.serenegiant.usb.encoder.MediaVideoEncoder;
import com.serenegiant.utils.FpsCounter;

import java.io.ByteArrayOutputStream;
import java.io.File;
import java.io.FileOutputStream;
[... 840 lines cut ...]
        String filePath = savebitmap(mTempBitmap, path);
        return filePath;
    }

    public String savebitmap(Bitmap bmp, String filePath) {
        ByteArrayOutputStream bytes = new ByteArrayOutputStream();
        bmp.compress(Bitmap.CompressFormat.JPEG, 60, bytes);
//        File f = new File(Environment.getExternalStorageDirectory()
//                + File.separator + System.currentTimeMillis() + "testimage.jpg");
        File f = new File(filePath);
        try {
            f.createNewFile();
            FileOutputStream fo = new FileOutputStream(f);
            fo.write(bytes.toByteArray());
            fo.close();
            filePath = f.getPath();
        } catch (IOException e) {
            e.printStackTrace();
        }

//        Bitmap bitmap = BitmapFactory.decodeFile(filePath);
        return filePath;
    }

}

GoldenStrawberry, 2018-06-20:
My camera supports Zoom D9: Zoom (Absolute)=enabled. I failed to set Zoom. How did you solve this problem?

shashankjamdade, 2018-06-26:
`/*
 *  UVCCamera
 *  library and sample to access to UVC web camera on non-rooted Android device
 *
 * Copyright (c) 2014-2017 saki t_saki@serenegiant.com
 *
 *  Licensed under the Apache License, Version 2.0 (the "License");
 *  you may not use this file except in compliance with the License.
 *   You may obtain a copy of the License at
 *
 *      http://www.apache.org/licenses/LICENSE-2.0
 *
 *   Unless required by applicable law or agreed to in writing, software
 *   distributed under the License is distributed on an "AS IS" BASIS,
 *   WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 *   See the License for the specific language governing permissions and
 *   limitations under the License.
 *
 *  All files in the folder are under this Apache License, Version 2.0.
 *  Files in the libjpeg-turbo, libusb, libuvc, rapidjson folder
 *  may have a different license, see the respective files.
 */

package com.serenegiant.usb.widget;

import android.content.Context;
import android.content.res.TypedArray;
import android.graphics.Bitmap;
import android.graphics.Canvas;
import android.graphics.Matrix;
import android.graphics.Paint;
import android.graphics.PointF;
import android.graphics.SurfaceTexture;
import android.hardware.Camera;
import android.hardware.camera2.CameraDevice;
import android.hardware.camera2.CaptureRequest;
import android.media.MediaRecorder;
import android.net.Uri;
import android.os.Build;
import android.os.Environment;
import android.os.Handler;
import android.os.Looper;
import android.os.Message;
import android.support.annotation.RequiresApi;
import android.util.AttributeSet;
import android.util.DisplayMetrics;
import android.util.Log;
import android.view.MotionEvent;
import android.view.ScaleGestureDetector;
import android.view.Surface;
import android.view.TextureView;
import android.view.View;

import com.jiangdg.libusbcamera.R;
import com.serenegiant.glutils.EGLBase;
import com.serenegiant.glutils.GLDrawer2D;
import com.serenegiant.glutils.es1.GLHelper;
import com.serenegiant.usb.IZoomCallback;
import com.serenegiant.usb.UVCCamera;
import com.serenegiant.usb.encoder.IVideoEncoder;
import com.serenegiant.usb.encoder.MediaEncoder;
import com.serenegiant.usb.encoder.MediaVideoEncoder;
import com.serenegiant.utils.FpsCounter;

import java.io.ByteArrayOutputStream;
import java.io.File;
import java.io.FileOutputStream;
import java.io.IOException;

/**
[... 838 lines cut ...]
        return filePath;
    }

    public String savebitmap(Bitmap bmp, String filePath) {
        ByteArrayOutputStream bytes = new ByteArrayOutputStream();
        bmp.compress(Bitmap.CompressFormat.JPEG, 60, bytes);
//        File f = new File(Environment.getExternalStorageDirectory()
//                + File.separator + System.currentTimeMillis() + "testimage.jpg");
        File f = new File(filePath);
        try {
            f.createNewFile();
            FileOutputStream fo = new FileOutputStream(f);
            fo.write(bytes.toByteArray());
            fo.close();
            filePath = f.getPath();
        } catch (IOException e) {
            e.printStackTrace();
        }

//        Bitmap bitmap = BitmapFactory.decodeFile(filePath);
        return filePath;
    }

}
`

jp1017, 2018-07-17:
Cooooooooooooooooooooooooooool

janoonaj, 2019-01-31:
How can you implement this class? it is using some unexisting code (ie IZoomCallback class which does not exist on the project). 

Actually, even the package is wrong, com.serenegiant.usb.widget; instead of com.serenegiant.widget.;

Am I missing something? 

thanks

heaveninfosoft, 2020-06-24:
@shashankjamdade   

hello how i can i build app for dual usb camera can support hd and fhd can  you help me???